FortySevenEffects/arduino_midi_library #30: sendPitchBend problem/bug with DefaultSettings

bastenhoor, 2015-02-27:
Hi,

When using:
struct MySettings : public midi::DefaultSettings
{
   static const bool UseRunningStatus = false; // Messes with my old equipment!
};
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial2, midi, MySettings);

I get an error when trying to sendPitchBend:

C:\Program Files (x86)\arduino\Arduino\libraries\MIDI/MIDI.hpp:282:57: error: 'Toto' is not a member of 'MySettings'
     const int value = inPitchValue * MIDI_PITCHBEND_MAX * Settings::Toto;

My guess is that  MIDI_PITCHBEND_MAX is not yet added to the DefaultSettings section and
Toto is a typo for Todo.

Franky47, 2015-02-27:
Ah, thanks for finding that !
I guess I used this method to test the settings overriding system, and left it there ever since.. oops.

closed 2015-02-27 by Franky47

bastenhoor, 2015-02-27:
Hi Francois,
Your welcome, thank you for creating and maintaining the great MIDI library! ;-)
I have a question, I've got a Yamaha WX5 midi breath/wind controller and recently received pressure sensors from Freescale.I'm trying to build my own wind controller. The MIDI data contains 'CC Breath' data (see attachment).Can I create this data with your library?
Thanks,Bas
Date: Fri, 27 Feb 2015 05:52:35 -0800

Franky47, 2015-02-28:
Sorry, I can't see your attachment, replying from email to GitHub does not seem to handle attachments..

Anyway, for breath control it looks like CC2/34 would be your best shot:
http://www.somascape.org/midi/tech/spec.html#ctrlnums

commit 3eb65db1 (2015-02-27): Fixed #30