FortySevenEffects/arduino_midi_library #32: Midi for USB - Need testers

NicoHood, 2015-04-19:
Hi there!
I guess you all use Midi for the Serial ports. I've implemented a USB-Core for Arduino Leonardo/Micro (witht he help of rkistner) to use Midi with your PC as well. If someone is interested in testing this or even combining it with this library, feel free to message me about any results.

There is not much documentation yet, since I just want to check if the basic recognition is working and people actually want to use this. Then I could develop it further and also improve the API and integration. It is just a quick startup to ask you if it work, since I cannot test the Host side at the moment.

Information/discussion can be found here (scroll down a bit)
https://github.com/NicoHood/HID/issues/2

Thx
Nico

Franky47, 2015-04-19:
Thanks Nico, that looks interesting !
I had good results recently with Dean Camera's LUFA library (but without the Arduino core), so I'll see what I can do with your code as well.

NicoHood, 2015-04-19:
I was also thinking of adding a 16u2 USB-Midi - Serial Firmware with Lufa. Which simple does a MidiUSB - Serial bridge instead of CDCSerial(USB) - Serial Bridge.

For this bridge one could use your library to communicate with this firmware maybe.

And for the 32u4 you have the IDE integration (not possible for the 16u2 via Arduino IDE at the moment due to code structure etc).

I think the API needs to be rewritten, but thats not a big deal. The main problem I have is no host test environment.

Franky47, 2015-04-19:
I think I've seen a 16u2 MIDI USB/Bridge hack somewhere to enable the library to work as a native MIDI USB (with the limitation that you need to use an external programmer to flash the main MCU). It was ages ago and I can't remember where it was though..

The trick for the 32U4 is to have a combination of both MIDI & CDC for application & firmware update without the use of an external programmer. I haven't investigated on "composite" USB applications yet.

Franky47, 2016-11-05:
Closing this as merging with [`arduino-libraries/MIDIUSB`](https://github.com/arduino-libraries/MIDIUSB) is planned (see arduino-libraries/MIDIUSB#2)

closed 2016-11-05 by Franky47