FortySevenEffects/arduino_midi_library #35: MIDI.sendControlChange(???);

sqrtmo, 2015-07-31:
Hello I am trying to achieve something like midi keypad with arduino and distance sensors.
I have four fixed notes and four distance sensors. I want to send a note with velocity( value coming from distance sensor ) to max/msp. I am using MIDI/MOCO for LUFA firmware. 

The problem is that somehow it changes the positions of the components of the midi package. 
In max I have displayed:
1. MIDI channel
2. Control number
3. Control value
Somehow and sometime I have control number on the place that should be control value???
I have tried with newer and older arduino versions. Where could be the problem?

I am attaching my patch. Can anyone help me? 
Thanks in advance.

#include <MIDI.h>
#include <FIRFilters.h>
MIDI_CREATE_DEFAULT_INSTANCE();

#define a (0.07)

filters<int> f0;
filters<int> f1;
filters<int> f2;
filters<int> f3;
filters<int> f4;
filters<int> f5;
filters<int> f6;

void setup()
{
  delay(500);
  MIDI.begin();
  delay(10);
  pinMode( 6, OUTPUT);
  pinMode( 9, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
  
  pinMode(13, OUTPUT);
  delay(10);
  digitalWrite( 13, HIGH );
}

void loop()
{ 
  uint16_t sen0 = f0.EMA( analogRead( A0 ), a );
  analogWrite( 11, (sen0/4) );
  playNote0( sen0 / 2 );
//----------------------------------------------------

  uint16_t sen1 = f1.EMA( analogRead( A1 ), a );
  analogWrite( 10, (sen1>>2) );
  playNote1( sen1 >> 1 );
//----------------------------------------------------

  uint16_t sen2 = f2.EMA( analogRead( A2 ), a );
  analogWrite( 9, (sen2>>2) );
  playNote2( sen2 >> 1 );  
//----------------------------------------------------  

  uint16_t sen3 = f3.EMA( analogRead( A3 ), a );
  analogWrite( 6, (sen3>>4) );
  playNote3( sen3 >> 1 );  
//----------------------------------------------------   

  uint16_t sen4 = f4.EMA( analogRead( A4 ), a );
  analogWrite( 6, (sen4/4) );
  playNote4( sen4 >> 1 );  
//----------------------------------------------------  

  uint16_t sen5 = f5.EMA( analogRead( A5 ), a );  
  analogWrite( 6, (sen5>>4) );
  playNote5( sen5 >> 1 );  
//----------------------------------------------------  

  uint16_t sen6 = f6.EMA( analogRead( A6 ), a );  
  analogWrite( 6, (sen6>>4) );
  playNote6( sen6 >> 1 );  
//----------------------------------------------------   

  delay( 10 ); //??  
}

//-----------PLAYNOTES----------------

#define midi_channel (1)                    

const byte notes[7] = { 45, 60, 80, 120, 32, 36, 43 };

void playNote0( uint16_t sen )
{ 
  static boolean flag = 0;
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[0], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[0], 0, midi_channel);
        flag = 0;
      }
  }
}

void playNote1( uint16_t sen )
{ 
  static boolean flag = 0;
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[1], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[1], 0, midi_channel);
        flag = 0;
      }
  }
}

void playNote2( uint16_t sen )
{ 
  static boolean flag = 0;  
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[2], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[2], 0, midi_channel);
        flag = 0;
      }
  }
}

void playNote3( uint16_t sen )
{ 
  static boolean flag = 0;
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[3], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[3], 0, midi_channel);
        flag = 0;
      }
  }  
}

void playNote4( uint16_t sen )
{ 
  static boolean flag = 0;
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[4], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[4], 0, midi_channel);
        flag = 0;
      }
  }  
}

void playNote5( uint16_t sen )
{ 
  static boolean flag = 0;
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[5], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[5], 0, midi_channel);
        flag = 0;
      }
  }  
}

void playNote6( uint16_t sen )
{ 
  static boolean flag = 0;
  if( sen > 0 )
  {
    MIDI.sendControlChange(notes[6], sen, midi_channel);   
    flag = 1;
    } else {
      if( sen == 0 && flag == 1 ){
        MIDI.sendControlChange(notes[6], 0, midi_channel);
        flag = 0;
      }
  }  
}

ricardogsilva, 2015-12-06:
HI @sqrtmo 

I was facing a similar problem just now (while trying out the arduino midi library). I got it working by changing the default baud rate. Adjust your code so that it looks like this:
```
MIDI.begin();
Serial.begin(115200);
```

Franky47, 2015-12-07:
The recommended way to change the default baud rate is to override the default settings, as shown:
```c++
struct MySettings
{
    static const long BaudRate = 115200;
};

MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);

void setup()
{
    MIDI.begin();
}
```
@ricardogsilva's solution is also valid, but ends up calling Serial.begin twice.

sqrtmo, 2015-12-09:
Thanks a lot for the replays! Actually I saw that the problem goes away if I reset the arduino with reset button, so that was my solution. But it is good to know for future projects.

Thanks again!

closed 2016-04-09 by Franky47