FortySevenEffects/arduino_midi_library #36: default instance doesn't compile

snerk4000, 2015-09-02:
after spending an entire day trying to get midi out to work on 32u4 based boards, it appears i needed to update to use it, but now nothing compiles and i get this: 

Arduino: 1.0.5 (Mac OS X), Board: "Arduino Micro"
MIDI_Basic_IO:7: error: expected constructor, destructor, or type conversion before ';' token

i've updated all the library files, and am using the file in the examples.  is there anything ELSE that needs to be changed to make this work?

Franky47, 2015-09-03:
Hi,

Sorry, I'm on holiday away from my computer until the end of the month, so
I'll have a look when I'm back.

Keep me posted if you figure out a way out of your problem.

Best regards,
François

Franky47, 2016-10-06:
Closed due to inactivity, and examples building fine on Leonardo / OS X / Arduino 1.6.

closed 2016-10-06 by Franky47