FortySevenEffects/arduino_midi_library #37: Handle RPN / NRPN CC messages

nick83ola, 2015-09-23:
I need to implement Sending/receiving Midi NRPN /RPN
https://en.wikipedia.org/wiki/NRPN
http://www.philrees.co.uk/nrpnq.htm
http://www.2writers.com/eddie/TutNrpn.htm

What is the best way to include them in library???

How I can write a callback for NRPN handling??

void sendNRPN(uint8_t channel, uint16_t parameter, uint8_t value) {
	MIDI.sendControlChange( 99, (parameter >> 7) & 0x7F, channel);
	MIDI.sendControlChange( 98, (parameter & 0x7F), channel);
	MIDI.sendControlChange( 6 , value, channel);
}
void sendNRPN(uint8_t channel, uint16_t parameter, uint16_t value) {
	MIDI.sendControlChange( 99, (parameter >> 7) & 0x7F, channel);
	MIDI.sendControlChange( 98, (parameter & 0x7F), channel);
	MIDI.sendControlChange( 6,  (value >> 7) & 0x7F, channel);
	MIDI.sendControlChange( 38, (value & 0x7F), channel);
}

void sendRPN(uint8_t channel, uint16_t parameter, uint8_t value) {
	MIDI.sendControlChange(101, (parameter >> 7) & 0x7F, channel);
	MIDI.sendControlChange(100, (parameter & 0x7F), channel);
	MIDI.sendControlChange(6,   value, channel);
}
void sendRPN(uint8_t channel, uint16_t parameter, uint16_t value) {
	MIDI.sendControlChange(101, (parameter >> 7) & 0x7F, channel);
	MIDI.sendControlChange(100, (parameter & 0x7F), channel);
	MIDI.sendControlChange(6,   (value >> 7) & 0x7F, channel);
	MIDI.sendControlChange(38,  (value & 0x7F), channel);
}

Franky47, 2015-09-23:
RPN/NRPN are not natively implemented in the library because it would make the parser much more complex, where not many people are using them in the end.

To receive them, you can add a little state machine in a handleControlChange callback that would recompose RPN/NRPN and call your own handlers.

starfishmod, 2015-11-09:
I certainly would love to be able to send and receive RPN and NRPN as I have a lot of equipment that rely on this (unfortunately).

At least sending would be nice although I will use the above code for the time being

Franky47, 2015-11-09:
I'll see what I can do.

Franky47, 2016-10-06:
I've pushed an (untested) [example](https://github.com/FortySevenEffects/arduino_midi_library/tree/master/res/Examples/MIDI_RPN_NRPN) that handles RPN & NRPN reading, I'm working on adding the send methods to the library core.

As you can see in the example, it's not trivial: RPN & NRPN reading requires a bit of a state machine / parser to handle the CC stream, and potentially separate state data (of the values of the RPN / NRPN of interest, as we don't want to try and read them all, it would not fit on in RAM).

commit 96374a1f (2016-10-06): Added RPN/NRPN sending facilities.
Needs doc, closes #37.
closed 2016-10-06 by Franky47