FortySevenEffects/arduino_midi_library #39: Problem with sendPitchBend

mpass0, 2016-04-05:
Hello! I'm trying to use arduino midi library to send signals from arduino to some musical software on my computer. I'm a beginner about this, so my first attempt is to send some noteOn/noteOff and PitchBend messages. I'm using a button for noteOn/noteOff and a potentiometer for pitchbend. Also, I read signals on the computer through the software MIDI-OX.
This is the code i've written:

#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();

int Button=LOW; int PrevButton=LOW;
int ButtonPin=10;

int note=60; int velocity=127; int channel=1;

int potVal=0;

void setup()
{
  MIDI.begin(MIDI_CHANNEL_OFF);
  pinMode(ButtonPin,INPUT);
}

void loop()
{
  Button=digitalRead(ButtonPin);
  potVal=analogRead(A0);
  int value=map(potVal,0,1023,-8192,8191);

  if (Button==HIGH && PrevButton==LOW){ // when i start pressing the button
    MIDI.sendNoteOn(note,velocity,channel);
  }
  if (Button==LOW && PrevButton==HIGH){ // when i release the button
    MIDI.sendNoteOff(note,0,channel);
  } PrevButton=Button;

  if (Button==HIGH && PrevButton==HIGH){ // only when i'm pressing the button
    MIDI.sendPitchBend(value,channel);
  }
   delay(200);
}

And here's what MIDI-OX receives:
[image]

What i don't understand is that after each pitchbend message there are lots of unwanted noteOn messages that, i think, are wrong. In fact, the status is 90 (as a noteOn), but data1 and data2 are the same as in the pitchbend (or similar because i was moving the potentiometer). 

Am i making any mistake in the code, or have i change or add other commands?

Thanks in advance.

Franky47, 2016-04-06:
Can you try replacing
```MIDI_CREATE_DEFAULT_INSTANCE();```
by
```c++
struct MySettings : public midi::DefaultSettings
{
  static const bool UseRunningStatus = false;
}
// Replace Serial by the serial port you're using for MIDI if not the default one.
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);
```

I'd like to rule out a running status issue, but one other thing that could happen here is bouncing of the button (electrical interference that cause digitalRead to oscillate between high and low when pressing or releasing a button for a short time), even though you use a timer of 200ms to slow down readings. There are lots of good libraries that handle debouncing very well, have a look on the Arduino Playground.

mpass0, 2016-04-07:
Thanks for your answer! However I solved the problem by replacing the function
`MIDI.sendPitchBend(value,channel);`
by the couple of functions
`MIDI.sendPitchBend(value,channel);`
`MIDI.sendNoteOff(0,0,0);`

Now it seems to work fine!

Franky47, 2016-04-09:
Looks like you're not the only one with this kind of issue, #41 looks similar. I'll try to replicate your case and see if I can solve this.

Thanks for the report !

mink99, 2016-04-10:
If you have a "flickering" button, that changes rapidly between high and low situations might occur that you miss one button - up or down.
To have less jitter on buttons, I often use pinmode(button, input_pullup) and then have button pressed = low, and button open = high. A further precaution would be to memorise whether a note on has been sent, and erase this flag on note off. Then, whenever you want to issue a note on, and still have a running note (flag is set) you can send an "emergency" note off.

Franky47, 2016-10-06:
Just as a remark, what are you using between the Arduino MIDI output and MIDI-OX (hardware and software) ?
Closing for now as this looks like a case of unhandling of Running Status.

closed 2016-10-06 by Franky47