FortySevenEffects/arduino_midi_library #41: sendControlChange(); generating NoteOn messages

IainD92, 2016-04-09:
Hey, I'm trying to use an Arduino Uno to create a device to turn PitchBend messages into CC messages (I have a sequencer that doesn't accept pitch bend, but uses a CC value for pitch shifting).

When I use MIDI.sendControlChange at the end, I instead get a load of NoteOn messages:
[http://i.imgur.com/wbpfSxC.png](url)

Here's the sketch I've written, am I doing anything stupid? Thanks in advance!
`#include <MIDI.h>

//constants
#define PITCHBENDMAX 8191
#define PITCHBENDMIN -8192

//variables
int CCPitch = 64;

MIDI_CREATE_DEFAULT_INSTANCE();

void setup()
{  
  MIDI.begin(MIDI_CHANNEL_OMNI);
  MIDI.setHandlePitchBend(TranslatePitchBend);
  MIDI.turnThruOff();
}

void loop()
{
  MIDI.read();
}

void TranslatePitchBend(byte channel, int bend)
{
  CCPitch = 1+ 127* float(
            (bend - PITCHBENDMIN)/
                float(PITCHBENDMAX-PITCHBENDMIN));
                
  MIDI.sendControlChange(80, CCPitch, channel);
}

`

Franky47, 2016-04-09:
You're not the only one with this kind of issue, I'll have a look and see if I can find and resolve this bug.

Thanks for reporting !

Franky47, 2016-04-09:
I wasn't able to reproduce this (although I use a Leonardo, but I don't think it matters):
[image]

However, I'm considering implementing feature request #40, which would let you remap PitchBend messages to CC80 without losing thru for all other messages.

IainD92, 2016-04-09:
That would be very useful! Very much looking forward to 4.3

Thanks for trying - in the meantime I've opted to send the 3 bytes like this and everything works:
```c++
void TranslatePitchBend(byte channel, int bend)
{
  CCPitch = 1+ 127* float(
            (bend - PITCHBENDMIN)/
                float(PITCHBENDMAX-PITCHBENDMIN));
                
 // MIDI.sendControlChange(80, CCPitch, channel);
  Serial.write(byte(175+channel));
  Serial.write(byte (80));
  Serial.write(byte(CCPitch));
}
```

Franky47, 2016-10-06:
It could be that your receiving device simply does not handle Running Status.

What hardware were you using to capture the MIDI stream coming out of the Arduino ?

Franky47, 2016-10-17:
There are many possible explanations for your NoteOn issues, one of which (after your driver not supporting Running Status) being that the stream started before the driver was ready, hence causing it to not receive the first status byte, but only data bytes. Interpreting them as NoteOn could be a default value for the Running Status implementation of the driver.

I'm closing this for now as I was not able to reproduce it on my tests (`sendPitchBend` sends correct `PitchBend` messages).

closed 2016-10-17 by Franky47