FortySevenEffects/arduino_midi_library #42: How to use pass MIDI pointer reference into a class?

trevor-sonic, 2016-04-18:
Sorry for this question but I couldn't find anywhere any info about it.
If we want to use MIDI.sendNote in a class/lib we need to pass the instance and/or pointer to the class but I couldn't find any example about it. If it's wrong place please delete my message.

closed 2016-04-18 by trevor-sonic

mink99, 2016-04-21:
Hi,

try this:
this is to create the midi interface as a **class member**
```

#include <MIDI.h>
class ThisClassHasAMidiPortAsMemberAndAFarTooLongName
{
  public:
    ThisClassHasAMidiPortAsMemberAndAFarTooLongName() {}
    void begin()
    {
      MIDI.begin(4);
    }
    
  private:
    midi::MidiInterface<HardwareSerial> MIDI = midi::MidiInterface<HardwareSerial>((HardwareSerial&)Serial1);
};

ThisClassHasAMidiPortAsMemberAndAFarTooLongName a;
void setup() 
{
  a.begin();  // to enable MIDI.BEGIN
}
void loop() {
  // put your main code here, to run repeatedly:
}

```
this is to create the midi interface **pointer as a class member**

```
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();
class ThisClassHasAMidiPortAsPointerAndAFarTooLongName
{
  public:
    ThisClassHasAMidiPortAsPointerAndAFarTooLongName() {}
    void begin(midi::MidiInterface<HardwareSerial> *pM)
    {
      pMIDI = pM;
    }
  private:
    midi::MidiInterface<HardwareSerial> *pMIDI;
};
ThisClassHasAMidiPortAsPointerAndAFarTooLongName a;
void setup()
{
  MIDI.begin();
  a.begin(&MIDI);
 
}
void loop() {
}

```

there is a bug in the gcc compiler that will not let you use that macro MIDI_CREATE_DEFAULT_INSTANCE(); within a class declaration.
therefore you will have to explicitely call the constructor.

took me a while to find out, you owe me a beer....

regards Kris

supported by ArtGenossen
underpowered by mink

Gesendet: Montag, 18. April 2016 um 12:26 Uhr

schlimme-gegend, 2017-01-23:
I just tried to implement the solution with the **pointer as a class member** and failed to compile. I suppose this happens because the **midi::MidiInterface<HardwareSerial>** does not provide a default constructor.
The error I get when my class' constructor is being compiled is:

`no matching function for call to 'midi::MidiInterface<HardwareSerial>::MidiInterface()'`

Simply adding a ctor that did nothing to MIDI.h and MIDI.hpp resulted in errors beyond my understanding of c++ , so this will not result in a pull request.

Franky47, 2017-01-23:
To reply to the original request:

You'll want to create a single instance of the library, then pass references to objects created later on. The macro can be used, but for the sakes of type consistency, using a `typedef` for the reference type is safer.

Such a pattern would look like this:
```c++
#include <MIDI.h>

typedef midi::MidiInterface<HardwareSerial> MidiInterface;

class MidiWorker
{
public:
    MidiWorker(MidiInterface& inInterface)
        : mMidi(inInterface)
    {
    }

public:
    void begin()
    {
        mMidi.begin();
    }
    void process()
    {
        mMidi.read();
    }

private:
    MidiInterface& mMidi;
};

// --

MidiInterface MIDI(Serial1);

MidiWorker worker(MIDI);

void setup()
{
    worker.begin();
}

void loop()
{
    worker.process();
}
```

@schlimme-gegend : The library stores a reference to the serial port object being used, therefore requires it to be initialized in ctor. Modifying the library to include a default ctor should therefore be limited to your specific use, where you would explicitly initialize the reference to that of your serial port of choice.

schlimme-gegend, 2017-01-23:
I was kind of sure that there is a good reason for the non existence of a default ctor ;) and then found myself overwhelmed when digging into your code. Thank you very much for pointing out a solution, I'll see that I get my stuff to work.

michaelkrone, 2017-01-29:
I also had the requirement to pass the `midi::MidiInterface` instance to other classes. Based on @Franky47 code, the following type definition is usefull when the default instance is used:
```c++
    typedef midi::MidiInterface<HardwareSerial, midi::DefaultSettings> MidiInterface;
```