FortySevenEffects/arduino_midi_library #45: Multiple Input Channels

masl123, 2016-07-01:
Hi,
I like the library, but there is one Problem I am having with it. You can not listen to more than one Input Midi Channel at a time. This is a serious limitation. My suggestion would be, to use a **Bitmask** instead of the Channel number to set the Channels to listen to. (So if bit 1 and 2 are set, you listen to Channels 1 and 2) At a later time in the "User Code", the User of the lib can sort the messages.

There are also two bugs in the Code, one might want to Fix:
1. in the inputFilter method in Midi.hpp:
  the Argument "inChannel" is not used, but there is "mInputChannel" used instead. (You can not specify the Channel to filter if you call the method)

2. in the thruFilter method in Midi.hpp:
  the Argument "inChannel" is not used, but there is "mInputChannel" used instead. (You can not specify the Channel to filter if you call the method)

Franky47, 2016-07-05:
The bit mask is a good suggestion, so far you could use Omni to listen to all channels and implement the filtering within your callbacks, but this approach sounds better.

You can open a PR with your code so I can review it.

Thanks !

Franky47, 2016-07-11:
For the record, the bug in the input filter was already fixed in 2197975.

franky47, 2021-11-03:
Upkeep: closing this issue due to lack of activity.

To listen to multiple channels, use the Omni mode and filter channels in the callbacks:

```c++
void handleNoteOn(midi::Channel channel, byte note, byte velocity)
{
  if (channel == 1 || channel == 2)
  {
    // todo: Implement me
  }
}

void setup()
{
  MIDI.begin(MIDI_CHANNEL_OMNI);
  MIDI.setHandleNoteOn(handleNoteOn);
}
```

closed 2021-11-03 by franky47