FortySevenEffects/arduino_midi_library #46: Can we have a callback example that reads SysEx?

bjosephs, 2016-07-05:
Hello,

I'm using a older version of the library and would like to try out the newer version to see if an issue I'm having is fixed (SysEx terminator F7 bytes read out of of memory as 0).  I can't get code to compile in Arduino IDE 1.6.7 with the teenseyduino add-in.  The release page for the new version says:

"SysEx Callback
The SysEx callback has a new signature:
void handleSysEx(byte* data, unsigned size);"

But I don't understand what this means, nothing seems to have changed in the wiki regarding sysex call backs.  This would compile and run using version 3.2:

`#include <MIDI.h>

void setup() {
  Serial.begin(115200);
  MIDI.begin(MIDI_CHANNEL_OMNI);
  MIDI.setHandleSystemExclusive(handleSysEx);
  Serial.println("MIDI SysEx Test");
}

void loop() {
  MIDI.read();
}

void handleSysEx(byte* array, byte array_size)
{
  for (int i = 0; i < array_size; i++)
  {
    Serial.print(array[i], HEX);
    Serial.print("\t");
  }
  Serial.println();
}
`

But not with 4.2.  I've tried adding the MIDI_CREATE_DEFAULT_INSTANCE(); but the   MIDI.setHandleSystemExclusive(handleSysEx); line causes and invalid conversion error.  I can't figure out what the correct syntax is to call my HandleSysEx function.

Franky47, 2016-07-05:
You need to replace the `byte array_size`in the signature of your callback by `unsigned array_size`.

The signature was changed to allow messages larger than 256 bytes to be parsed and received (as a byte would overflow).

bjosephs, 2016-07-05:
Wow, thanks.  I can't believe that was all it was.  I must have been overthinking it because the it was right there in the error message...  As an aside the terminating byte now reads as F7, so definitely worth the time.  Thanks for this amazing library.

closed 2016-07-05 by bjosephs