FortySevenEffects/arduino_midi_library #47: SysEx messages longer than 127 bytes are ignored by MIDI.setHandleSystemExclusive()

bjosephs, 2016-07-05:
Hello,

I have code that was working with version 3.2, receiving and Serial.print'ing long SysEx messages (tested up to 200 bytes).  I see that version 4.2 includes a bug fix for messages longer than "byte" length but now my callback function is either ignoring or not receiving messages that are longer than 127 bytes.    In MIDI.cpp there is a function called decodeSysEx, but no hints on where to use it.  Do I need to copy that function code into my sketch?  Can it be run within my callback function or does it need to follow MIDI.read?  For reference, here is my call back function:

`void handleSysEx(byte* array, unsigned array_size)
{
  for (int i = 0; i < array_size; i++)
  {
    Serial.print(array[i], HEX);
    Serial.print("\t");
  }
  Serial.println();
}`

Brian

Franky47, 2016-07-05:
Hi Brian,

`decodeSysEx` is useful only if the data you're receiving is plain old binary data (where the MSB can be 1), encoded into SysEx data (where the MSB has to be 0), with `encodeSysEx`.

However, I don't think the issue is here, I'll have a look in the code to see if there might be a bug in the parsing of the SysEx size. I assume then your `array_size` is always <= 127 when the callback is invoked ?

bjosephs, 2016-07-05:
Oh, OK.  Glad to hear that.  The array can be smaller.... the messages from this device are typically 8 or 9 bytes, though one is hard coded at 200 and another can vary from a few dozen to a few hundred depending on settings.  In case it helps here is what I'm receiving:

http://wiki.fractalaudio.com/axefx2/index.php?title=MIDI_SysEx

I can send you some data too if you need it.  I adjusted the settings on the device so I could see that when a message is below 127 it "arrives", change the settings so that message needs to include a few more bytes and it stops chowing up.

Thanks again!
Brian

bjosephs, 2016-07-06:
FYI, version 4.1 has the F7 > 0 issue fixed AND does not seem to have the same issue with the SysEx length.

Franky47, 2016-07-07:
Interesting.. I'll have a look.

Thanks !

sanotronics, 2016-08-08:
Hello!

I am having this exact same problem.

Is there a quick workaround?

sanotronics, 2016-08-08:
Update: On line 73 of the file midi_Settings.h there is a variable called SysExMaxSize, which I found by default on 128.

I changed it and now it can receive bigger messages.

Franky47, 2016-10-10:
@bjosephs, I'm writing unit tests to try and reproduce your issue, what did you mean by:

> version 4.1 has the F7 > 0 issue fixed

bjosephs, 2016-10-10:
I referring to a bug in library v3.2 that caused me to start looking at 4.1 and 4.2.  I brought that bug up in a different thread so I really should have given it some context here.  In that library version the terminating byte of a SysEx message, though being sent as an 0xF7, was reading out of memory as 0x00.

Franky47, 2016-10-29:
Hi @bjosephs,

I don't see any issues with the parser or the callbacks, I've tested it with 210 bytes of data and the message is received correctly in the callback, as long as the `SysExMaxSize` setting is set to a value large enough to accommodate your largest expected SysEx message. Maybe your problem is here (as it was for @sanotronics) ?

You can [override the default setting](https://github.com/FortySevenEffects/arduino_midi_library/wiki/Using-custom-Settings) with a larger value:
```c++
struct MySettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.
};

// Create a 'MIDI' object using MySettings bound to Serial1.
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);
```

I'll close this for now, please let me know if you have any updates.

closed 2016-10-29 by Franky47

eclab, 2016-10-30:
I have a different need.  I have an extremely memory constrained application that will need to read/write some 400 bytes of Sysex data.  The existing library as far as I can tell must allocate a buffer and load the whole thing before handing it to me.   How can I avoid this?  I had hoped I could have things work as follows:

INPUT.  A handler is called when a small number of Sysex bytes are received.  You're told if the bytes are the start, middle, or end of a start message.  This would allow me to load bit by bit.

OUTPUT.  I can send the start, muddle, and end of a Sysex message rather than all of it at once.

I'd also need to decode 7->8 bits (and encode) in these small chunks.  Ideas?  Or am I way off base?