FortySevenEffects/arduino_midi_library #49: PitchBend problem

nishiratavo, 2016-08-13:
Hi, 
I'm trying to use sendPitchBend to send a float number, but I got a error with that function it. It says that "Toto" is not a member of "midi::DefaultSettings". I have searched the library and couldn't find that variable anywhere. Is it a problem with my code or a bug in the library ?

My code

```
#include <MIDI.h>
#include <Wire.h>
#include <SparkFun_MMA8452Q.h>

float ym;
float n;

MIDI_CREATE_DEFAULT_INSTANCE();

MMA8452Q accel;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(57600);
  accel.init(SCALE_8G, ODR_200);
  MIDI.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
    for(n = 0; n<30;n++)
    {
    accel.read();
    ym = ym + accel.cy;
    }
    ym = ym/n;
    if(MIDI.read())
    {
    MIDI.sendPitchBend(ym,1);
    }
} 
```

Franky47, 2016-08-19:
Oh wow, yes, it's the remains of an old test, I'll fix that.

In the mean time, you can replicate the function this way:
```c++
const int pitchBend = int(ym * float(MIDI_PITCHBEND_MAX));
MIDI.sendPitchBend(pitchBend, 1);
```

The library method uses double, not sure if they're at all supported on AVR platforms, but it seems a bit overkill for a pitch bend control, float would be much more efficient (especially since the value is then encoded on 14 bits).

Franky47, 2016-08-19:
Fixed in 55f91d54aac28931e6a946229a18fdccea19c336.

closed 2016-08-19 by Franky47