FortySevenEffects/arduino_midi_library #51: Problem with receiving more than one SysEx message

sanotronics, 2016-09-30:
Hello!

I am having trouble to receive multiple sysex messages on an Arduino UNO.

I am using a self-made multiplatform SysEx configurator that sends 207 messages packets. This had to be done because of an OSx limitation that prevents me from sending SysEx messages greater than 256 bytes (you can find several forum posts with this issue on google).

I am receiving this data and writing it on the EEPROM. 

After this, I am reading it to check what was written, and I am having several issues, always with the second and further packets. Sometimes it writes ok, sometimes there are bytes displacements... I can't find any patterns to this.

I tried sending the same message using Bome's SendSX (https://www.bome.com/products/sendsx) to send 3 packets, but separate in time, instead of one after another, and it worked perfectly.

I tried compressing the 3 packets into one large message, and it also worked.

I tried with this same software sending the 3 packets one after another with SendSX and I get similar results as with my software.

This is to tell you that I think the writing and reading into the EEPROM seems to work fine, but when I try packets that arrive very close in time, something is not working.

I don't know if it's a library issue, or a HIDUINO issue... I will post this in both githubs.

If I could get help with this it'd be awesome.

I attach the sketch I am using to write and read the EEPROM (switch between both commenting the WRITE_EEPROM define), the library that formats the data into the eeprom, and the sysex data I am testing with.

If you need further info on the issue or something is not clear, let me know.

Thanks in advance!

[data sysex.txt](https://github.com/FortySevenEffects/arduino_midi_library/files/503226/data.sysex.txt)
[Sysex1.zip](https://github.com/FortySevenEffects/arduino_midi_library/files/503221/Sysex1.zip)
[KilomuxSysEx.zip](https://github.com/FortySevenEffects/arduino_midi_library/files/503224/KilomuxSysEx.zip)

Franky47, 2016-09-30:
This could be a concurrency problem between receiving a sequence of SysEx messages at the MIDI rate and writing to the EEPROM.

What I would suggest is to implement a queue where you'd push your SysEx messages when received (don't forget to copy the array contents as they are overwritten when receiving new messages), and where you would pop them to write to the EEPROM when it becomes ready, until the queue is empty.

sanotronics, 2016-09-30:
Ok, I'll try and come back with results. I'm not entirely sure how to do it, but I'll give it my best.

Thanks

Franky47, 2016-10-13:
Closing for now, reopen with details if the issue persists after buffering. Thanks !

closed 2016-10-13 by Franky47

sanotronics, 2016-10-13:
ok, I tried buffering, but it's not a solution, since I can get messages up to 1024, in chunks of 200 bytes, so the buffer has to be 1024 bytes in order to accomodate the whole message.

The Arduino UNO has a 2 KB SRAM, and half of that for this buffer is too much. I changed the pc SysEx configurator to wait 1 second between each chunk of data, and now it works.

Not the perfect solution, but it does the job for the moment.

CaptainCredible, 2016-10-13:
I also have this error and solved it by programming the arduino to stop
everything it's doing as soon as it starts receiving sysex messages.  And
focus only on the sysex until it receives the sysex message telling it that
the programming is over.