FortySevenEffects/arduino_midi_library #52: Integrate with PluggableUSB for native USB MIDI

Franky47, 2016-10-06:
It seems easier now to integrate with native USB for communications, so writing an interface for the library that uses native USB as transport protocol could be investigated.

Also, see #32.

CaptainCredible, 2016-10-06:
Great news!

Franky47, 2016-10-12:
Good news: it's fairly simple to integrate with the MidiUSB Arduino library, there's still a bit of work to be done in [`midi_UsbTransport.hpp`](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/src/midi_UsbTransport.hpp), but I'm already able to receive and send NoteOn/NoteOff messages.

Example: [`MidiUSB.ino`](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/MidiUSB/MidiUSB.ino)

Bad news: the latency is terrible, but I was kind of expecting it. The issue here is that the wrapping around MidiUSB makes it double the work needed to receive a message.
ie: data arrives from USB bus, is pre-parsed by the MidiUSB library into `midiEventPacket_t` objects, then the MIDI library consumes it by re-transforming them into a serial stream.

In order to make this work efficiently, I'd have to re-code a large part of the MidiUSB library into the MIDI library, effectively creating a non-official duplicate.. I'll have to see with the Arduino team what they think of this approach.

dmadison, 2017-06-08:
Has there been any progress on this since last year? I've been doing a bit of work on a MIDI project and I'd like to eventually integrate both USB MIDI and serial MIDI on the same board.

marcosassis, 2018-03-07:
hey @franky47 , your library is great and beautiful, thanks very much for your work and attention, I really appreciate it

> Bad news: the latency is terrible, but I was kind of expecting it. The issue here is that the wrapping around MidiUSB makes it double the work needed to receive a message.

but this problem is huge, and in my opinion it's a design bad choice in the basis of the library (which is yours or MidiUSB's creator I don't know...)

I'm new on MIDI and USB development, but I've been studying some things (I'm using pro micro at the moment, but I intend to be as generic as possible in my works, with some nice options already out-of-the-box)

I almost used your library's interface to MIDIUSB in my project (and honestly... I don't know how to say this), but I can't figure out what do I have to do to write/send a single MIDI note on this. here my pro micro can play/send notes MIDI over MIDIUSB used directly (as in example MIDIUSB_write), but it doesn't works using midi::UsbTransport. and instead of trying to figure out what happened here and try to fix it **on the promise of a bad latency on final**, I'd prefer to study (more than I initially expected in this phase of my project) more MIDI and USB and re-write everything (maybe even including MIDIUSB functionality). 

but, anyway, it seemed more reasonable to try to open this channel with you, so we can exchange important information and different points of view of the same problem. 

if you see these, please feedback us about this issue. I hope with some collaboration, we can or chose one way to do this together, or at least greatly improve each approach we choose to solve this.

best regards

franky47, 2018-03-07:
Hi @marcosassis,

The `midi::UsbTransport` is the most naive way of connecting the current MIDI library (which works by parsing a byte stream) with the MIDIUSB library (which, as per the MIDI USB spec, works with packets), and was only made as an exploratory attempt to enable native USB capabilities. To be honest, it should probably not have ended up in the repository at all.

A more realistic approach to interfacing MIDI through both USB and Serial transports would probably require splitting the interface from the parser, having separate parsers for various transport types. The issue is that it may become less straightforward for beginner users to handle.

I'll have to check what options we have in the Arduino compilers to make this transparent to users.

marcosassis, 2018-03-11:
hello again @franky47 !

this is my attempt of 'proof of concept' for showing how i'd resolve this important issue:

https://github.com/marcosassis/gamepaduino/blob/master/src/midi_proto/midi_interface.h

(I'm only sending commands, as it's what I need _for now_.)

I had no previous experience with any MIDI at all (I'm not a musician), before I buy my pro micro and start to write code for it, not long ago. Of course I'm studying your code, and I'd like to thank you very much again for making this accessible. So I'm here to humbly share these experiments, in the hope to learn more in this matter.

this is `midi_interface` implemented for USB using MIDIUSB library:

https://github.com/marcosassis/gamepaduino/blob/master/src/midi_proto/midi_usb_interface.h

I'm using pro micro 16MHz (5V)

tested on (windows7 64) MIDI-OX and Ableton* with success (no latency perceptible)

*(this a friend of mine, Monty Cantsin, tested for me)

**this was one of the most funny things I ever did with arduino** =] and that's why:

https://github.com/marcosassis/gamepaduino/blob/master/examples/N64_usb_midi_debug/N64_usb_midi_debug.ino

this is a N64 controller (original) playing MIDI notes AND pitch-bending then (analog x axis)

(I'm also happy because I really like template programming, so please don't judge my tastes =S.) 

`midi_usb_interface` is used as

```
// this uses a 'interval map + tone' method for playing MIDI notes as buttons are pressed
typedef midi_instrument<N64_gamepad>  N64_midi;

// this pitch bends any midi_instrument passed as parameter (type and base object)
typedef pitch_wheel<N64_midi>         N64_bender;

// this is midi_interface implemented (only send for now) for MIDIUSB (library adapter)
using   meta::midi::midi_usb_interface;

// initialization of everybody
int8_t note_map[N64_gamepad::N_BUTTONS]={0,1,2,3,4,5,6,7,16,9,10,11,12,13,14,15};
midi_data_t                                      basetone=69; // A4
midi_usb_interface            midiusb;
N64_midi           p1midi(p1, midiusb, note_map, basetone);
N64_bender         p1bender(p1midi);
```

bender is read on loop (and automatically play MIDI notes and commands when button pressed/released)
and, **after all buttons->notes where processed**, only then, communication is flushed. so, loop is:

```
  p1midi.read();
  midiusb.flush();
```

pitch bend is crazy!! =D and it's a nice way to flood this MIDI communication and feel(as we're not measuring yet) the real time behavior of this thing. but  if one doesn't have a N64 controller, but luckily have a SNES controller, they can test this too:

https://github.com/marcosassis/gamepaduino/blob/master/examples/SNES_usb_midi_debug/SNES_usb_midi_debug.ino

franky47, 2018-11-06:
There is now a working implementation on the [`feat/4.4.0`](https://github.com/FortySevenEffects/arduino_midi_library/tree/feat/4.4.0) branch, you'll also need to get the latest code from `master` on [`MIDIUSB`](https://github.com/arduino-libraries/MIDIUSB) until the next release of MIDIUSB is published.

Turns out it was not a latency problem, but a broken implementation (only some messages were able to pass through). I now have very low latency and complete interfacing between the MIDI Library and USB MIDI.

Congrats on your game controller application, it sounds like a lot of fun !

closed 2018-11-06 by franky47

giovannibotta, 2019-01-30:
Will this branch get integrated back in the trunk?