FortySevenEffects/arduino_midi_library #65: Normalise default serial ports for common platforms

Franky47, 2016-11-04:
## Platforms currently in production

Name | Processor | Framework | Default HW Serial Port | USB support | Definition
-----|-----------|-----------|------------------------|-------------|-----------
Uno | ATmega328P | Arduino | Serial | no | `ARDUINO`
101 | ARCV2EM | Arduino | Serial1 | yes | ?
Pro | ATmega328P | Arduino | Serial | no | `ARDUINO`
Zero | SAMD21G18A | Arduino | Serial | yes | ?
Mega | ATmega2560 | Arduino | Shared? | no | ?
Teensy 2.0 | ATmega32U4 | Teensyduino | ? | yes | ?
Teensy++ 2.0 | AT90USB1286 | Teensyduino | ? | yes | ?
Teensy LC | MKL26Z64 | Teensyduino | ? | yes | ?
Teensy 3.2 | MK20DX256  | Teensyduino | ? | yes | ?
Teensy 3.5 | MK62FX512 | Teensyduino | ? | yes | ?
Teensy 3.6 | MK66FX1M0 | Teensyduino | ? | yes | ?

## Discontinued platforms

Name | Processor | Framework | Default HW Serial Port | USB support | Definition
-----|-----------|-----------|------------------------|-------------|-----------
Due | SAM3X8E | Arduino | Shared? | yes | `ARDUINO`
Leonardo | ATmega32U4 | Arduino | Serial1 | yes | ?
Teensy 3.0 | MK20DX128 | Teensyduino | ? | yes | ?
Teensy 3.1 | MK20DX256 | Teensyduino | ? | yes | ?

Franky47, 2016-11-04:
Relates to the following issues & PR:
- #11 - Default serial port is wrong on Leonardo
- #29 - USBCON is defined only for AVR
- #33 - Digging out definitions for default serial port (lost in the middle of other topics).
- #63 - Teensy handling.

Franky47, 2016-11-04:
When `SERIAL_PORT_HARDWARE_OPEN ` is defined, it points to the hardware UART that is available for external serial IO, which is what we're after:
```c++
// These serial port names are intended to allow libraries and architecture-neutral
// sketches to automatically default to the correct port name for a particular type
// of use.  For example, a GPS module would normally connect to SERIAL_PORT_HARDWARE_OPEN,
// the first hardware serial port whose RX/TX pins are not dedicated to another use.
//
// SERIAL_PORT_MONITOR        Port which normally prints to the Arduino Serial Monitor
//
// SERIAL_PORT_USBVIRTUAL     Port which is USB virtual serial
//
// SERIAL_PORT_LINUXBRIDGE    Port which connects to a Linux system via Bridge library
//
// SERIAL_PORT_HARDWARE       Hardware serial port, physical RX & TX pins.
//
// SERIAL_PORT_HARDWARE_OPEN  Hardware serial ports which are open for use.  Their RX & TX
//                            pins are NOT connected to anything by default.
```

lathoub, 2020-12-12:
Should the USB part of this table be moved over to https://github.com/lathoub/Arduino-USBMIDI?