FortySevenEffects/arduino_midi_library #66: Errors trying to override Max SYsex msg length

bossredman, 2016-11-09:
Hi,

I'm trying to override the Sysex Max length locally in my sketch as advised by adding the following code after `  #include <MIDI.h>:`

`  Struct MySettings : public midi::DefaultSettings`
`{
    static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.
};`

`// Create a 'MIDI' object using MySettings bound to Serial1.`
`MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);`

But I get the following errors:

**_C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:14:30: error: 'midi' has not been declared

   struct MySettings : public midi::DefaultSettings

C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:14:36: error: expected '{' before 'DefaultSettings'

   struct MySettings : public midi::DefaultSettings

C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:15:1: error: invalid type in declaration before '{' token

 {

C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:16:5: error: expected primary-expression before 'static'

     static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.

C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:16:5: error: expected '}' before 'static'

C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:16:5: error: expected ',' or ';' before 'static'

C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino:18:1: error: expected declaration before '}' token

 };_**

Fairly new to this so would appreciate any help pls.

Franky47, 2016-11-09:
Hum, that's odd, I'll give it a try tomorrow. Are you running v4.3.1 (latest release) ?

bossredman, 2016-11-10:
hi
Srtill on 3.2

bossredman, 2016-11-10:
Hi - updated to 4.2 and got this:

>
In file included from C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:8:0:

>C:\Users\Medion\Documents\Arduino\libraries\MIDI_Library\src/midi_UsbTransport.h:32:21: fatal error: MIDIUSB.h: No such file or directory

 >#include <MIDIUSB.h

>compilation terminated.

>Error compiling for board Teensy 3.6.

So I commented out `#include <midi_UsbTransport.h>`  - but now get this:

> In file included from C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:2:0:

>C:\Users\Medion\Documents\Arduino\libraries\MIDI_Library\src/midi_Defs.h:236:5: error: 'MidiInterface' in namespace 'midi' does not name a type

>    midi::MidiInterface<Type, Settings> Name((Type&)SerialPort);

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:31:5: note: in expansion of macro 'MIDI_CREATE_CUSTOM_INSTANCE'

>    MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);

>Error compiling for board Teensy 3.6.

When I tried to update to 4.3.1, the midi_library folder was empty - so reverted back to 4.2

bossredman, 2016-11-10:
UPDATE - Managed to get v4.3.1 installed but still getting the same errors as previous post.

Franky47, 2016-11-10:
Did you manage to solve your issue in the end ?

bossredman, 2016-11-10:
No - ref 

> UPDATE - Managed to get v4.3.1 installed **but still getting the same errors as previous post.**

Franky47, 2016-11-10:
It looks like the main class definition (`midi::MidiInterface`) was not found, it's located in `MIDI.h`, do you include that before using the `MIDI_CREATE_CUSTOM_INSTANCE` macro ?

bossredman, 2016-11-10:
Hi - yes I do:

`
#include <MIDI.h>
#include <midi_Defs.h>
#include <midi_Message.h>
    #include <midi_Namespace.h>
    #include <midi_RingBuffer.h>
    #include <midi_Settings.h>
    #include <midi_UsbDefs.h>
    //#include <midi_UsbTransport.h>`

bossredman, 2016-11-10:
Ref the first error (ie the one I get without the `#include <midi_UsbTransport.h>` line being commented out - should there be a MIDIUSB.h file included in teh library folder?

Franky47, 2016-11-10:
Ah ok, you don't actually need to include all the headers, only `#include <MIDI.h>` will do.

USB transport is a preview of native USB capability for compatible boards, that requires a dependency on the [`MIDIUSB`](https://github.com/arduino-libraries/MIDIUSB) library. It is not required for serial operations.

bossredman, 2016-11-10:
hmm OK - so I commented everything out appart from MIDI.h.
Get even more errors now:

> C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:11:32: error: 'midi' has not been declared

> struct MySettings : public midi::DefaultSettings

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:11:38: error: expected '{' before 'DefaultSettings'

>struct MySettings : public midi::DefaultSettings

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:12:1: error: invalid type in declaration before '{' token

 >{

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:13:5: error: expected primary-expression before 'static'

>static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:13:5: error: expected '}' before 'static'

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:13:5: error: expected ',' or ';' before 'static'

>C:\Users\Medion\Documents\Arduino\_Sketches_TEENSY_36_MFC\Midi_to_AXEFX_Teensy-36_rev63\Midi_to_AXEFX_Teensy-36_rev63.ino:15:1: error: expected declaration before '}' token

> };

>Error compiling for board Teensy 3.6.

Franky47, 2016-11-10:
Can I have a look at your sketch please ?

bossredman, 2016-11-10:
//Libraries
  #include <LiquidCrystal.h>
  #include <Bounce.h> //Buttons
  #include <elapsedMillis.h>
  #include <MIDI.h>
      
//    struct MySettings : public midi::DefaultSettings
//{
//    static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.
//    static const bool UseRunningStatus = true; // My devices seem to be ok with it.
//};
//    
//    // Create a 'MIDI' object using MySettings bound to Serial2.
//    MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);

Franky47, 2016-11-10:
I've tried your code with Teensy 3.6 v1.31-beta3 / Arduino IDE v1.6.12 / MIDI lib v4.3.1, and I have no issues building it.

Just to be sure, on Teensy 3.6 the hardware serial port is accessed through `Serial1`, right ?

bossredman, 2016-11-10:
Hi - tbh - not sure what you mean by "hardware serial port is accessed through Serial1."

I'm not teh most technical on this stuff & have kind of fudged my way through it all.

Is this serial port configurable? Should I be setting it somewhere/somehow.
As I dont believe I am doing that.

ANother thing - how can I confirm what version of teh library I have installed?
The release notes show it as 4.2 (11/06/2014)  but the Library Manager says I have 4.3.1 installed.
The midi.h file whilst has no version number listed - states the date as  *  @date       24/02/11

I downloaded & installed the library direct form the Arduino IDE.

I'm confused

Franky47, 2016-11-11:
On all Teensy boards, you have two serial interfaces: `Serial` which communicates with the computer through the USB port (USB serial), and `Serial1` which uses the hardware UART (RX and TX pins on the Teensy pinout) for communication with external devices.

So now, depending on what you want to do:
- Communicate with a hardware synth or MIDI machine: connect DIN jacks to the RX & TX pins, and use `Serial1`
- Communicate with the computed via USB: there is no official support of native MIDI USB yet, so you'll have to piggyback the MIDI messages on top of the serial comms. This is handled by 3rd party projects on the computer such as Hairless MIDI.

I see that your project file is named AxeFX, so option 1 would be most probable.

bossredman, 2016-11-11:
Hi - Many thanks fo your very helpful explanation.

Yes - you are correct, my project is Axefx 2 based. 
It is a DIY midi foot controller.
I am indeed using the Teensy's RX & TX pins (0 & 1) to provide Midi IN & Midi OUT for the Axefx & MFC via DIN jacks and approriate midi IN/OUT circuits.
So guess that is using `Serial1`.

I also use the 'Serial Monitor' function to help troubleshoot.
I'm guessing that therefore uses `Serial` via the Teensy's mini USB socket when I connect to my PC.
I also use the Teensy's mini USB socket for power (via PC when programming or a USB charger plug for std use).

At no point in my code do I configure/setup serial ports apart from `Serial.begin(115200);`

I've managed to get most of the features/functionality working OK.
But have rcently tried to read one particular sysex msg for reading the states of Effect Blocks (0x0E).
This has proved troublesome. 
Sometimes it works/sometimes not - depending where in the code I call it from.
Basically I reads OK if called from void loop().
But I can not get it to work if called from the void setup() block.

I used the Serial Monitor to list the bytes of the 0x0E sysex response (via Serial.print's) & it looks like the msg was over 300 bytes long. With the last 200 or so 0's.
This has drove me down the route of trying to override the Max SysEx message length & to this forum.

Last night I basically started again - ie - uninstalled Arduino IDE/ reinistalled (v1.6.12)/Teensyduino (v1.31)/midi library v4.3.1 -but still get errors when trying to run the msg length override code.

So I have a few more questions pls.
1] Do I need to be doing more to configure/setup the serial ports?
2] Do I really need to over ride the max SysEx msg length?
3] Do you know that the dates & versions in your latest v4.3.1 library files are not up to date & do not reference v4.3.1

Franky47, 2016-11-11:
Ok, so based on your needs, a basic setup would look like this:
```c++
#include <MIDI.h>

struct MySettings : public midi::DefaultSettings
{
    // override default settings here
};

MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

void setup()
{
    Serial.begin(115200);       // Start serial comms for debugging
    while(!Serial);             // Wait for the serial port to be initialized
    MIDI.begin();               // Start MIDI communications, will take care of Serial1.
    Serial.println("Ready");    // Print debug string
}

void loop()
{
    MIDI.read();
}
```

Regarding the size of the SysEx buffer, it should be as big as the biggest message you expect to receive. It does not have to be a power of two, so you can put just what you need there not to burn too much RAM. Even if the received message contains a lot of zeros, they are still part of the data and will count for the message size (ie: number of bytes for the whole message between/including 0xF0 and 0xF7).

The dates are the creation dates of those files, maybe I should just remove them.. The version number of the library can be found in either `library.properties` or `library.json`.

bossredman, 2016-11-11:
Thanks again.

This is drivng me crazy now :) .
Checked both library (.properties & .json) files & both indeed show v4.3.1.

I then created a new sketch ("Franky47_test" ) & literally copied **only** your above code into it.
Tried to compile/verify it & get this:

> Franky47_test:3: error: 'midi' has not been declared
 struct MySettings : public midi::DefaultSettings

>Franky47_test:3: error: expected '{' before 'DefaultSettings'
 struct MySettings : public midi::DefaultSettings

>Franky47_test:4: error: invalid type in declaration before '{' token
 {

>Franky47_test:8: error: 'Serial1' is not a type
 MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

>Franky47_test:8: error: 'MIDI' is not a type
 MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

>Franky47_test:8: error: expected constructor, destructor, or type conversion before ';' token
 MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

Franky47, 2016-11-12:
Ok so there must be something wrong with the library files that causes the namespace not to be recognized.

Can you turn on verbose compilation in the Arduino settings and paste the whole compiler output here please ?

bossredman, 2016-11-12:
Here you go .. thanks for your continued support,

>
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Medion\Documents\Arduino\libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180opt,keys=en-gb -ide-version=10612 -build-path C:\Users\Medion\AppData\Local\Temp\arduino_build_711513 -warnings=all -verbose C:\Users\Medion\Documents\Arduino\Franky47_test\Franky47_test.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Medion\Documents\Arduino\libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180opt,keys=en-gb -ide-version=10612 -build-path C:\Users\Medion\AppData\Local\Temp\arduino_build_711513 -warnings=all -verbose C:\Users\Medion\Documents\Arduino\Franky47_test\Franky47_test.ino
Using board 'teensy36' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Using core 'teensy3' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=131 -DARDUINO=10612 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_UNITED_KINGDOM "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3" "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\sketch\Franky47_test.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=131 -DARDUINO=10612 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_UNITED_KINGDOM "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI" "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\sketch\Franky47_test.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=131 -DARDUINO=10612 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_UNITED_KINGDOM "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI" "C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI\MIDI.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=131 -DARDUINO=10612 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_UNITED_KINGDOM "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI" "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\sketch\Franky47_test.ino.cpp" -o "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=131 -DARDUINO=10612 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_UNITED_KINGDOM "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI" "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\sketch\Franky47_test.ino.cpp" -o "C:\Users\Medion\AppData\Local\Temp\arduino_build_711513\sketch\Franky47_test.ino.cpp.o"
Franky47_test:3: error: 'midi' has not been declared
 struct MySettings : public midi::DefaultSettings

>Franky47_test:3: error: expected '{' before 'DefaultSettings'
 struct MySettings : public midi::DefaultSettings

>Franky47_test:4: error: invalid type in declaration before '{' token
 {

>Franky47_test:8: error: 'Serial1' is not a type
 MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

[... cut ...]
 MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

>Franky47_test:8: error: expected constructor, destructor, or type conversion before ';' token
 MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

>Multiple libraries were found for "MIDI.h"
 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI
 Not used: C:\Users\Medion\Documents\Arduino\libraries\MIDI_Library
Using library MIDI in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI (legacy)
'midi' has not been declared

Franky47, 2016-11-12:
I think this is a good lead:
```
Multiple libraries were found for "MIDI.h"
Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI
Not used: C:\Users\Medion\Documents\Arduino\libraries\MIDI_Library
Using library MIDI in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI (legacy)
```

bossredman, 2016-11-12:
Arh...

Looks like the version in 'C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\MIDI' is 3.2.

`*  @file		MIDI.h
 *  Project		MIDI Library
 *	@brief		MIDI Library for the Arduino
 *	Version		3.2
 *  @author		Francois Best 
 *	@date		24/02/11
 *  License		GPL Forty Seven Effects - 2011
 */`

Will it simply be a case of deleting/moving this folder & the IDE should then use the  library in 'C:\Users\Medion\Documents\Arduino\libraries\MIDI_Library', which is v4.3.1?

`name=MIDI Library
version=4.3.1
author=Forty Seven Effects
maintainer=Francois Best <francois.best@fortyseveneffects.com>
sentence=MIDI I/Os for Arduino
paragraph=Read & send MIDI messages to interface with your controllers and synths
category=Communication
url=https://github.com/FortySevenEffects/arduino_midi_library
architectures=*
includes=MIDI.h`

Franky47, 2016-11-12:
Yep ! Just remove the old version and the IDE should pickup the new one automatically.

bossredman, 2016-11-12:
Thanks.

I moved it to a seperate folder and now my test sketch  (Franky47_test.ino) complies without error.

I'll double check my original/main sketch later & confirm.

I wonder if this is also the cause of my other issue  - ref #67?

Thanks once again.

Franky47, 2016-11-12:
Ahh, yes, I did not realize you were the one who opened #67, so yes, it's probably related (see my comment on #67).

bossredman, 2016-11-12:
OK thanks very much - all is now working in my original sketch.
(Midi_to_AXEFX_Teensy-36_rev62\Midi_to_AXEFX_Teensy-36_rev62.ino).
Although I did have to make a further change to my code to accomodate the new Library (the type of arraySize):

From:        `void HandleSystemExclusive(byte *SysExArray, byte arraySize)     `
To:             `void HandleSystemExclusive(byte *SysExArray, unsigned arraySize)    `

All complie errors are gone & I can now overide SysExMaxSize (my original issue).

Looks like it was down to having muliple versions of the Midi library on my PC and my sketch kept picking up the old one.

[and yes issue #67 is now also resolved]

Thanks for all you help on this one.

closed 2016-11-12 by bossredman