FortySevenEffects/arduino_midi_library #67: SysEx 0xF7 not being read

bossredman, 2016-11-11:
Hi,

I've seen mentions of similar issues in earlier versions of the Midi Library (ie pre-v4.2) where F7's (ie msg terminating byte) were being read as 0's.
But that in v4.2 + it was fixed.

I now also seeing this issue but its not just with converting to 0.
I'm also seeing other values too.

I'm running a Teensy 3.6 with: Arduino IDE (v1.6.12) / Teensyduino (v1.31) / midi library v4.3.1.
Project is a midi foot controller for AxeFX2.

I've just noticed this whilst looking into another issue (ref: "Errors trying to override Max SysEx message length #66")  [https://github.com/FortySevenEffects/arduino_midi_library/issues/66](url)

Franky47, 2016-11-11:
Do you have an example message that was received incorrectly (but somehow parsed correctly as a SysEx message ?)

bossredman, 2016-11-11:
Here's one for the Preset Number - in this case it was preset 4.         

 ++++ PRESET-NUMBER SYSEX - 0x14  +++++
               4
0  =  F0
1  =  0
2  =  1
3  =  74
4  =  3
5  =  14
6  =  0
7  =  4
8  =  16
9  =  46

Response format (from Axefx Wiki):

    0xF0 sysex start 
    0x00 Manf. ID byte0 
    0x01 Manf. ID byte1 
    0x74 Manf. ID byte2 
    0x03 Model # 
    0x14 Function ID (0x14) 
    0xdd Preset Number bits 6-0 
    0xdd Preset Number bits 13-7 
    0xdd checksum 
    0xF7 sysex end

bossredman, 2016-11-11:
..and also one for Preset name - in this case "Stone Roses":

          ++++ PRESET-NAME SYSEX - 0x0f  +++++
               Stone Roses         
0  =  F0
1  =  0
2  =  1
3  =  74
4  =  3
5  =  F
6  =  53
7  =  74
8  =  6F
9  =  6E
10  =  65
11  =  20
12  =  52
13  =  6F
14  =  73
15  =  65
16  =  73
17  =  20
18  =  20
19  =  20
20  =  20
21  =  20
22  =  20
23  =  20
24  =  20
25  =  20
26  =  20
27  =  20
28  =  20
29  =  20
30  =  20
31  =  20
32  =  20
33  =  20
34  =  20
35  =  20
36  =  0
37  =  0
38  =  12
39  =  33

Response format:

    0xF0 sysex start 
    0x00 Manf. ID byte0 
    0x01 Manf. ID byte1 
    0x74 Manf. ID byte2 
    0x03 Model # 
    0x0F Function ID (0x0F) 
    0xdd null-terminated string byte0 
    0xdd byte1 
    ... 
    0x00 null character 
    0xdd checksum 
    0xF7 sysex end

Franky47, 2016-11-12:
Thanks, so you definitely get a SysEx response from the library (MIDI.read() returns true and `MIDI.getMessageType() == midi::SysEx`, or end up in a handleSysEx callback), and your buffer looks like this ?

If you get the message recognized as SysEx, and as there is no particular way of knowing a generic SysEx length other than waiting for 0xF7, it means it was correctly received, but somehow dropped by the parser. I'll run some tests and I'll get back to you.

bossredman, 2016-11-12:
Ref    
> and your buffer looks like this ?

They were copies from the serial monitor output. 
FYI - Here's my code that gets them.
`
```
void HandleSystemExclusive(byte *SysExArray, byte arraySize)     //Gets called if a SysEx msg is recieved after the MIDI.read() command is called.
{
if (SysExArray[5] == 0x14)          //If Read a Preset Number Sysex
    {
      int sizear = MIDI.getSysExArrayLength();
      
      Serial.println("          ++++ PRESET-NUMBER SYSEX - 0x14  +++++");
      BANK = (MIDI.getSysExArray()[6]);
      PRESET_NUM = (MIDI.getSysExArray()[7]);
      MIDI.sendSysEx(8,Preset_Name_Sysex,true);
      Serial.print("               "); Serial.println(PRESET_NUM);
      for (int v=0; v<sizear; v++)
        {
            Serial.print(v);  Serial.print("  =  ");  Serial.println(SysExArray[v], HEX);
        }
    }

if (SysExArray[5] == 0x0f)          //If Read a Preset Name Sysex
     {
         int sizear = MIDI.getSysExArrayLength();
         
         Serial.println("          ++++ PRESET-NAME SYSEX - 0x0f  +++++"); 
         for(int i=0;i<20;++i)  //was 31
          {
            pname[i] = MIDI.getSysExArray()[i+6];
          }
        Setup_Compl = true;
        LCD_Print();
        Serial.print("               "); Serial.println(pname);
    
        for (int v=0; v<sizear; v++)
        {
            Serial.print(v);  Serial.print("  =  ");  Serial.println(SysExArray[v], HEX);
        }
      
     } 
}   
```
`

Franky47, 2016-11-12:
I see you're using a `byte arraySize` in the callback signature, are you using an old version of the library ?

The latest version (v4.3.1) is available [here], there were a lot of bug fixes since the signature was changed to unsigned (to handle large SysEx larger than 255 bytes).

Try and update and let me know how it goes.

bossredman, 2016-11-12:
Hi.

The resolution to issue #66 has also fixed this issue.
Looks like it was down to having muliple versions of the Midi library on my PC and my sketch kept picking up the old one.

I can now read F7's.

As you pointed out - I did have to make a further change to my code to accomodate the new version of the Library (the type of arraySize):

From: void HandleSystemExclusive(byte *SysExArray, byte arraySize)
To: void HandleSystemExclusive(byte *SysExArray, unsigned arraySize)

Cheers.

closed 2016-11-12 by bossredman