FortySevenEffects/arduino_midi_library #68: Getting only F0 F7 at midi monitor using sendSysEx function

sanotronics, 2016-11-25:
Hi!

I am trying to send a SysEx message from my Arduino UNO, and when I check on a MIDI monitor what is that I am sending, I see this:

```
 TIMESTAMP IN PORT STATUS DATA1 DATA2 CHAN NOTE EVENT               
       848  9  --    240 Buffer:     2 Bytes    System Exclusive      
  SYSX: F0 F7
      1849  9  --    240 Buffer:     2 Bytes    System Exclusive      
  SYSX: F0 F7
      2852  9  --    240 Buffer:     2 Bytes    System Exclusive      
  SYSX: F0 F7
      3851  9  --    240 Buffer:     2 Bytes    System Exclusive      
  SYSX: F0 F7
      4851  9  --    240 Buffer:     2 Bytes    System Exclusive      
```

and so on... The monitor is MIDI Ox, and I see the same thing with MIDI Monitor in a Mac computer.

I am using the following code on an Arduino UNO, which only sends SysEx, but the same thing happens when I send in a code that does other stuff:

```
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE()

byte i = 0;

void setup() {
  // put your setup code here, to run once:
  MIDI.begin(MIDI_CHANNEL_OMNI); MIDI.turnThruOff();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  byte dumpOkMsg[7] = {'Y', 'T', 'X', i, 0};
  MIDI.sendSysEx(5, dumpOkMsg, false);
  i++;
  i %= 128;
  delay(1000);
}
```
I am using HIDUINO firmware on the Atmega16u2.

Same problem [here on the forum](https://forum.arduino.cc/index.php?topic=274811.0) and [here on HIDUINO's GH](url)

Has anybody here came across this before?

koenschepens, 2016-11-25:
The problem is in hiduino not supporting sysex properly. I switched to another hid firmware (forgot the name and I'm traveling now so I don't have access). Just so you know :)

caryfitzhugh, 2017-03-27:
@koenschepens I'm about ready to write a PR against hiduino to support outgoing sysex commands - but if you had a different firmware that already does it, maybe you could share it? Thanks!

sanotronics, 2017-03-28:
mocoLUFA supports outgoing and ingoing (64 byte limit) sysex messages and dual boot 
https://github.com/kuwatay/mocolufa

koenschepens, 2017-03-29:
@caryfitzhugh I switched to mocoLUFA as well.

caryfitzhugh, 2017-03-30:
Thanks... just switched over. a little finicky, but seems like I got it to
work.  thanks everyone!

closed 2017-10-30 by franky47