FortySevenEffects/arduino_midi_library #70: Arduino Due, IDE 1.6.14, Library not accepted

EnfieldRider, 2016-12-11:
Hello Friends,
I'm having problems with this library.
The IDE throws error 'MIDI' was not declared ...
#include <MIDI.h> is recognized, but nothing more.
The library vers 4.3.1 is installed in common way from zip. All is running on a MacBook.

Code (part):
<<<
#include <MIDI.h>

void setup() {
  MIDI.begin(MIDI_CHANNEL_OMNI);
  Serial.begin(57600);
  Serial.println("MIDI Input Test");
}

unsigned long t=0;

void loop() {
  int type, note, velocity, channel, d1, d2;
  if (MIDI.read()) {                    // Is there a MIDI message incoming ?
    byte type = MIDI.getType();
...
>>>
Errors:
<<<
/Users/me/Documents/Arduino/MIDItest/MIDItest.ino: In function 'void setup()':
MIDItest:4: error: 'MIDI' was not declared in this scope
   MIDI.begin(MIDI_CHANNEL_OMNI);
   ^
/Users/me/Documents/Arduino/MIDItest/MIDItest.ino: In function 'void loop()':
MIDItest:13: error: 'MIDI' was not declared in this scope
   if (MIDI.read()) {                    // Is there a MIDI message incoming ?
       ^
MIDItest:16: error: 'NoteOn' was not declared in this scope
       case NoteOn:
            ^
... and so on
<<<

Doed anybody know the reason why?

Franky47, 2016-12-11:
You need to add the following line to create the `MIDI` instance object:
```c++
#include <MIDI.h>

MIDI_CREATE_DEFAULT_INSTANCE(); // Add this to create the MIDI instance.

void setup()
{
    MIDI.begin();
    // ...
}
```

Also, you'll need to prefix MIDI definitions such as `NoteOn` and other message types with the `midi` namespace, as such:
```c++
if (MIDI.getType == midi::NoteOn) // Instead of just NoteOn
{
}
```

EnfieldRider, 2016-12-12:
Thank you very much.
I'd took an old example file and did not understand the confusing error messages from IDE.
I realized the problem after using the program body after "getting started" and filled it up step by step.

closed 2016-12-12 by EnfieldRider