FortySevenEffects/arduino_midi_library #76: SPI Interface instead of Serial

Jorgen-VikingGod, 2017-05-10:
Hi guys,
is there a way to build a "simple" SPI brigde for this library?

for my next project I want to use the ESP32 as main MCU, but it do not have USB driver. So I will use a small MCU for doing the USB stuff.
I want do show up my "device" as MIDI device over USB; the small MCU has to pass the MIDI data over SPI to the ESP32 and the ESP32 should fill data in the lib, so I can use the callbacks and send methods.

Has anyone an idea how to do this as fast as possible, without changing to much of the lib?
Thx

Franky47, 2017-05-10:
That should be fairly easy, as the library uses a template-based interface with whatever underlying hardware layer is used for transport.

You could write a wrapper class that has the following signature:
```c++
class SpiMidiTransport
{
public:
    void begin(long inBaudRate);    ///< Initialize (ignore baudrate for SPI)

public:
    int available() const;          ///< Return the number of bytes available in the SPI RX buffer.
    void write(byte inData);        ///< Write a byte to SPI TX
    byte read();                    ///< Read a byte from the SPI RX buffer
};
```
Then you would use it like this:
```c++
#include <MIDI.h>

SpiMidiTransport spiTransport; // Create wrapper instance
MIDI_CREATE_INSTANCE(SpiMidiTransport, spiTransport, spiMidi); // Pass it to the library

void setup()
{
    spiMidi.begin();
}

void loop()
{
    spiMidi.read();
}
```

Jorgen-VikingGod, 2017-05-10:
@Franky47 wow this is awesome - such a great and fast reply.
Thank you very much for your support - I will try to add this to my project

closed 2017-05-11 by Jorgen-VikingGod