FortySevenEffects/arduino_midi_library #77: Custom SysEx length setting works up to 63 bytes, not more.

Razoumihine, 2017-05-12:
Good day  :)
Not sure the issue is due to the library. 
I'm using library V 4.3.1 on an Arduino MEGA 2560, and I'm stuck on an issue where the maximum size of a receivable SysEx message is 63bytes. 
Although all defaults setting are at 128 bytes, I'm still limited at 63. This means:
If I set the SysExMaxSize = 16 , then it limits to 16 bytes
If I set the SysExMaxSize = 58 , then it limits to 58 bytes
If I set the SysExMaxSize = 64 or beyond , then it limits to 63 bytes, whatever is the max size setting. 
I've no idea how to fix this.... Maybe you could help?
Here is the code I use, if this can help...

`//With MIDI OUT plug to MIDI IN, this code sends to MIDI OUT a SisEx message 
//with increasing size at each time a button is pressed. The code receives this 
//Sysex message to the MIDI IN prints it to serial monitor.  

/////////////////////////MAIN SETTINGS////////////////////////////
#include <MIDI.h>
#include <Keypad.h>

struct MySettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 128 ; 
    // Accept SysEx messages up to xxx bytes long.
//but works only bellow 64 bytes
};

// Create a 'MIDI' object using MySettings bound to Serial1.
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial1, MIDI, MySettings);

//////////////////////////////KEYPAD//////////////////////////////////
const byte ROWS = 4; //four rows
const byte COLS = 4; //four columns
char keys[ROWS][COLS] = {
  {'a','b','c','d'},
  {'e','f','g','h'},
  {'i','j','k','l'},
  {'m','n','o','p'}
};
byte rowPins[ROWS] = {41, 39, 37, 35}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {27, 29, 31, 33}; //connect to the column pinouts of the keypad

Keypad keypad = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS );

int i = 60; // first SysEx size setting

/////////////////////////////////////////////////HANDLE SYSEX/////////////////////:
void handleSysEx(byte* inData, unsigned inSize)
{
    Serial.println(inSize);
}

////////////////////////////////////////////SETUP////////////////////////
void setup()
{
  Serial.begin(9600);
  MIDI.begin();
  MIDI.setHandleSystemExclusive(handleSysEx);
  MIDI.turnThruOff();
}

///////////////////////////////////////////LOOP///////////////////////////////////:
void loop()
{
byte Message[i];     
//*************************SEND MESSAGE******************************************    
  char key = keypad.getKey();
  
  //Si une touche du clavier est enfoncée
  if (key != NO_KEY)
  { 
       for (int j = 0; j<i; j++)
           {
            Message[j]=0;
           }
        
    MIDI.sendSysEx(i, Message, false); // envoie le message sysex
    Serial.print("SysEx  sent: ");
     //Boucle pour écrire dans le moniteur série le message SysEx envoyé
     for (int j = 0; j<i; j++)
      {
      Serial.print(Message[j],HEX);
      Serial.print(" ");
       }
    Serial.println("");
    i++; 
   }
   
//*************************RECEIVE******************************************       

  MIDI.read();
}
`
Cheers, 
Lionel

Razoumihine, 2017-05-13:
Finally solved. 
Not sure why, but, I did a small code to try to find the bug: This was a loop sending a Sysex message with higher length step by step, up to 63 bytes it worked, and more it didnt. 
I assume now it is the sent message that was limited, and not the receiving one, cause with the real MIDI instrument, the arduino finally receives longer messages. 
Is the maximum length for SysEx sending is the same setting as for SysEx receiving in the library?

Franky47, 2017-05-13:
Sending SysEx (as well as any other message type) uses the Arduino's TX buffer, which could be 64 bytes long.

The receiving part is different, in the sense that two "buffers" are involved. The RX buffer, which buffers incoming bytes between calls of MIDI.read(), and the SysEx buffer which recomposes those bytes read from MIDI.read into a meaningful SysEx message (as long as it is within the boundaries defined in the settings).

So if you need to send large SysEx, you should implement some kind of packet splitting system to make sure your message (or split parts of it) can fit into the Arduino TX buffer.

closed 2017-05-16 by Franky47