FortySevenEffects/arduino_midi_library #80: Midi Out on ESP8266 UART1

tonywestonuk, 2017-06-29:
ESP8266 based boards (Lolin / witty/ a few others), have an annoying serial debug output on UART 0, causing problems in midi-out.  

Is it possible to use UART1 TX (accessible by Serial1.begin() ) for Midi Out ? -  Apparently, UART1 doesn't have RX on 8266 boards..... but, I only need it for midi out.

tonywestonuk, 2017-06-29:
I found it!  :-D
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, midi1);

closed 2017-06-29 by tonywestonuk

Renlong2020, 2021-02-05:
> 我找到了！:-D
> MIDI_CREATE_INSTANCE（HardwareSerial，Serial1，midi1）;

Hello, although the problem has been in the past 3 years, I would like to ask, can your code running on 8266 output MIDI signals correctly?

tonywestonuk, 2021-02-05:
Crikey !  I forgot I even posted this.

Yes, it did output midi signals correctly on the 8266. Though, I have upgraded to use ESP32 now.

Renlong2020, 2021-02-05:
> key！我什至忘了贴这个。
> 
> 是的，它确实在8266上正确输出了MIDI信号。不过，我现在已经升级为使用ESP32。

Can you share your code, run on 8266, can it output MIDI signal correctly?
My code is correct on UNO, but not on 8266.
Simply send noteOn in a simple loop.

Also, this is my code:
Can not output correctly on 8266, but UNO can.
#include <MIDI.h>
MIDI_CREATE_INSTANCE(HardwareSerial, Serial, MIDI);
void setup() {
   // put your setup code here, to run once:
   //Serial.begin(31250);
   MIDI.begin(MIDI_CHANNEL_OMNI); //Start midi
   MIDI.turnThruOff(); //Prevent MIDI loop
}

void loop() {
   // put your main code here, to run repeatedly:
   MIDI.sendNoteOn(1,127,1);
   delay(1000);
}