FortySevenEffects/arduino_midi_library #84: MTC Output problem

edwar64896, 2017-09-05:
I have just started using your library along with HIDUINO - I am intending to build an LTC to MTC converter on a MEGA2560R3 Platform. Good results so far - I have an interrupt driven LTC reader working well using Timer4 on the 2560 and I have built sketches that allow me to send MTC into the device and display it on a 16x2 LCD.

What I have been struggling with is getting MTC out of the unit. I am using a second timer to drive the MTC output events, which I am scheduling as 4 events per frame of LTC received. I"m using a moving average to determine the spacing of the MTC events in clock pulses based on the total clock time elapsed per frame of LTC decoded and displayed.

I have determined that the MTC timer is actually being invoked, however any attempt to send an MTC Quarter Frame is failing. I'm not seeing any MTC packets on the destination device at all, although other midi messages can be seen.

I am wondering whether there is an issue with serial speed somewhere as the midi note-on messages I am seeing appear to be arriving quite slow, compared with the frequency they are being sent. I am wondering if there is serial setting somewhere that needs to be altered to increase the speed of transmission. I can see HIDUINO toggling the TX LED as the MTC packets are sent to the HIDUINO device (16u2 on the MEGA) but they appear to get no further and are not detected on the midi monitor I am using (OSX).

Any pointers/hints would be much appreciated.

Franky47, 2017-09-05:
The slow NoteOn messages can indicate that you're swarming the TX buffer, with more MTC data going in that can be pulled out and transmitted (at the standard MIDI rate of 31250 baud).

You could try and raise the baudrate, but only if your receiving device is configured to align on a different rate (like a computer's serial line), usually all MIDI equipment sticks to the standard baudrate.

Now as for why the MTC packets are not visible, maybe it comes from the difference between Quarter Frame messages, which require 8 messages to make a full information, vs MTC System which uses a System message to encode the whole data. Which one are you using ?

For the record, I used this documentation while building the library:
http://www.somascape.org/midi/tech/spec.html#rusx_mtc

edwar64896, 2017-09-06:
QuarterFrame messages. 

void sendMTC() {

  switch (MTCIndex)
  {
    case 0:
      toSend = (f & 0xF);
      break;

    case 1:
      toSend = ((f & 0xF0) >> 4);
      break;

    case 2:
      toSend = (s & 0xF);
      break;

    case 3:
      toSend = ((s & 0xF0) >> 4);
      break;

    case 4:
      toSend = (m & 0xF);
      break;

    case 5:
      toSend = ((m & 0xF0) >> 4);
      break;

    case 6:
      toSend = (h & 0xF);
      break;

    case 7:
      toSend = ((h & 0xF0) >> 4 ); // 0x70 = 24 fps // 0x72 = 25 fps // 0x74 = 30df fps // 0x76 = 30 fps
      break;
  }

  MIDI.sendTimeCodeQuarterFrame(MTCIndex++, toSend);

  if (MTCIndex > 7)
    MTCIndex = 0;
}

edwar64896, 2017-09-06:
So my current strategy is to have two timers running. Timer 4 is exclusively used for decoding the LTC stream. Timer 5 is used for sending MTC. I am determining the number of clock ticks between full received frames of LTC, then dividing this by 4 and using this as the interval at which I send MTC messages.I'm also resetting MTCIndex each time a Zero frames value comes in from LTC, so that the MTC message sequence starts afresh each time a new timecode second starts.

franky47, 2018-03-10:
Hi @edwar64896, did you manage to solve your problem ? I have a bit of time between jobs to do a round of QA on the library, so let me know if I can help.

edwar64896, 2018-03-11:
hi @franky47 there was some success I managed to achieve with the library, which was around implementing MTC. It's a bit of a hack and you could probably do a better job, so I will upload the diff for you to do with what you will.
[dualMoco.c.diff.txt](https://github.com/FortySevenEffects/arduino_midi_library/files/1800023/dualMoco.c.diff.txt)

What's there works and I have implemented an LTC to MTC translator which amazingly manages to synch protools quite reliably.

franky47, 2018-10-11:
Your diff is about LUFA, not sure what to do with that..

closed 2018-10-11 by franky47