FortySevenEffects/arduino_midi_library #85: handleNoteOff ch got wronkly sometimes

billyjoker, 2017-11-01:
Hi, i have some problems when send quickly note on and note off. The handleNoteOff callback receives a wrong channel number:

handleNoteOff Note Off1: ch=5, pitch=9, note=9 -> OK
handleNoteOff Note Off1: ch=5, pitch=9, note=9 -> OK
handleNoteOff Note Off1: ch=5, pitch=0, note=0 -> OK
handleNoteOff Note Off1: ch=5, pitch=1, note=1 -> OK
handleNoteOff Note Off1: ch=1, pitch=9, note=9  -> What??? I sent the same ch that before this one...

Any ideas???

tmarti, 2017-11-25:
Had the same problem with Control Change messages.

By the moment, workarounded it by downgrading library to v4.2.

Try if this helps you.

billyjoker, 2017-11-27:
Hi @tmarti , i'll do in your way and i'll tell you something. My temp solution was to send the note off 4 repeated times, to reach the note off of my affected channel.  

Thanks for your time!!

tmarti, 2018-04-24:
Hi @billyjoker, did the workaround work for you?

closed 2020-05-17 by franky47