FortySevenEffects/arduino_midi_library #86: Issue receiving Real Time messages over USB

connortreacy, 2017-11-27:
Hi @franky47, I'm trying to use your library with my Arduino Leonardo to build a USB midi interface. I've been struggling to receive Real Time messages, **clock** specifically. 

I'm using Ableton as a DAW, and my Leonardo shows up as a midi device, where's it's configured for **Track**, **Sync** and **Remote** output midi messages. 

I've compiled and uploaded a basic Arduino sketch to my Leonardo, which should echo to Serial the value of getType for every midi message received:

```
#include <MIDI.h>

#if defined(USBCON)
#include <midi_UsbTransport.h>

static const unsigned sUsbTransportBufferSize = 16;
typedef midi::UsbTransport<sUsbTransportBufferSize> UsbTransport;

UsbTransport sUsbTransport;

MIDI_CREATE_INSTANCE(UsbTransport, sUsbTransport, MIDI);

#else // No USB available, fallback to Serial
MIDI_CREATE_DEFAULT_INSTANCE();
#endif

void setup() {
    Serial.begin(115200);
    while (!Serial);
    MIDI.begin();
}

void loop() {
    if(MIDI.read()) {
      Serial.println(MIDI.getType(), HEX);
    }
}
```

However, my Serial monitor only shows Note On, Note Off and `B0` messages, which appear to be sent when I press Stop in Ableton. 

I can eavesdrop on what my Leonardo receives, using [MIDI Monitor](https://www.snoize.com/MIDIMonitor/), and here I see that the Leonardo receives Start, Clock and Stop events too:

```
20:36:44.928	To Arduino Leonardo	Start		FA
20:36:44.928	To Arduino Leonardo	Clock		F8
20:36:44.949	To Arduino Leonardo	Clock		F8
20:36:44.969	To Arduino Leonardo	Clock		F8
...
20:36:45.886	To Arduino Leonardo	Clock		F8
20:36:45.907	To Arduino Leonardo	Clock		F8
20:36:45.912	To Arduino Leonardo	Stop		FC
20:36:45.912	To Arduino Leonardo	Control	1	B0 7B 00
```

Any idea why the library isn't offering up these messages?

https://github.com/FortySevenEffects/arduino_midi_library/blob/780923567609d385e6e566144a7fe609ebdb1f1a/examples/MidiUSB/MidiUSB.ino#L44

krizroring, 2018-07-14:
Created a PR to fix the problem: https://github.com/FortySevenEffects/arduino_midi_library/pull/100

franky47, 2018-10-11:
It's interesting, as there is a handler for 1-byte messages (Clock, Start, Stop etc) in the USB transport layer, I'll have a look.

As a side note, `B0 7B 00` is the ControlChange message for AllNotesOff.

mink99, 2018-10-31:
Isn’t the serial and the USB port internally the same on the leonardo ? So usb communication and serial.print cannot be used the same time (like on the uno)

franky47, 2018-10-31:
@mink99 if so, then issue #103 would be in the same situation.

However, some of the MIDI-USB library examples use serial debugging, see:
https://github.com/arduino-libraries/MIDIUSB/blob/master/examples/MIDIUSB_write/MIDIUSB_write.ino

franky47, 2018-10-31:
Ok I've done a quick test with Ableton Live, and indeed I can't seem to receive Start/Clock/Stop messages, I'll investigate.

franky47, 2018-10-31:
@mink99 to get this out of the picture, I can confirm it's possible to use serial debugging along with reception of MIDI input messages on a Leonardo.

A few things to remember though:
- Wait for the serial port to connect with `while(!Serial);` (waiting for USB enumeration I suppose).
- Open the serial monitor on the Arduino IDE, otherwise the `while` loop will never break.

As for the true nature of this issue, I believe I found the error in my code, I was checking for the USB packet header to contain the actual 1-byte message for Real Time messages, but the header appears to be 0x0F for every type of Real Time message, and the actual message being stored in the first data byte:
[image]

I'll get a fix on this today.

mremusic, 2018-11-06:
@franky47 can you maybe tell us a way for a temporary fix? i tried a few things but wasn't able to solve it...

franky47, 2018-11-06:
Yes, I've got a working fix on the [`feat/4.4.0`](https://github.com/FortySevenEffects/arduino_midi_library/tree/feat/4.4.0) branch, you'll also need to get the latest code from `master` on [`MIDIUSB`](https://github.com/arduino-libraries/MIDIUSB) until the next release of MIDIUSB is published.

closed 2019-02-22 by franky47