FortySevenEffects/arduino_midi_library #89: Problem using MIDI and serial monitor printouts with Teensy

kschaffer, 2017-12-18:
This could very well be something stupid that I am doing, but I am running into a problem that I lose the ability to use serial printouts for diagnostics when I invoke the begin() method for my MIDI interface with Teensy 3.2.  The problem and code to reproduce it are in the Teensy forum:  [https://forum.pjrc.com/threads/48491-problem-using-MIDI-library-and-serial-monitor-at-the-same-time](https://forum.pjrc.com/threads/48491-problem-using-MIDI-library-and-serial-monitor-at-the-same-time)

Is there a way around this?

franky47, 2017-12-18:
I see nothing wrong with your code, you should be able to use two serial ports at once, one for MIDI and one for printout debugging. Since you passed `Serial2` as the argument of the constructor, the MIDI Library will only care about this object and not cause side effects on other serial ports, so there could be something else going on here.

kschaffer, 2017-12-18:
Thanks a lot for your quick reply, and sorry to bother you with this if it has nothing to do with your code!  I will keep trying things out and see where it ends up.  Much appreciated.

closed 2018-01-18 by franky47