FortySevenEffects/arduino_midi_library #90: sendNoteOn/Off working only with hardcoded parameters

nomadic68, 2018-02-18:
Hi, i am using an Arduino with MIDI shield to interface 2 commercial instruments (so I have no access to their software). Basically the Arduino act as a "man in the middle" to translate the SysEx messages from the controlling instrument to noteOn/Off for the controlled instrument.
My SysEx callback looks like this:

void SysExMsg(byte message[], unsigned int size)
{
  byte code = getCode(message);
    
  switch(code)
  {
     case 0x0a:
         MIDI.sendNoteOn(Note1, ON, ch1);
         break;
     case 0x0b:
         MIDI.sendNoteOn(Note2, ON, ch2);
         break;
     case 0x1a:
         MIDI.sendNoteOff(Note1, ON, ch1);
         break;
     case 0x0b:
         MIDI.sendNoteOn(Note2, OFF, ch2);
         break;
     // etc etc
   }
}

where the Note1, Note2, ch1, ch2 etc are external variables (actually, constants) mapping the instruments to be controlled.
My problem is that this way the note on/off are not sent at all, while if I hardcode the parameter values into the sendNoteOn/Off calls everything works fine.
The Note1, Note2, ch1, ch2 etc variables are actually symbolic names for the desired sound effect and are also used elsewhere in my code, so losing the symbolic name is a big disadvantage.
I tried defining the symbols as: #define, global const, global var, local const, local var; nothing.
I also looked into the library source code but didnt find anything, maybe i overlooked something.
I wonder if it could be the Arduino IDE messing up the compilation instead.

Thanks and regards,
nomadic68

franky47, 2018-02-19:
Sounds like an `enum` would do what you need ?

nomadic68, 2018-03-09:
Hi all, problem solved! I found that i was inadvertently passing invalid channel number (something like channel instead of table[channel]). My bad, i should have named variables differently to avoid this confusion (now i did).
Thanks and regards

closed 2018-03-09 by nomadic68