FortySevenEffects/arduino_midi_library #94: Using MIDI library in a project made of multiple source files

nomadic68, 2018-03-09:
Hi all, my MIDI Arduino software has grown to the point that i decided to organize the source code in a set of multiple source and header files. Besides of that, i moved from Arduino IDE to Atmel Studio 7, in order to have stricter compiler checks which results in writing proper C++ code (Arduino IDE is way too much forgiving for professional use).
My problem is that i call MIDI_CREATE_DEFAULT_INSTANCE() in file1.cpp but i cannot figure out the correct extern declaration in the other source/header files and compilation aborts.

What is the correct extern declaration for MIDI object for my other source header files?

Thanks and regards

franky47, 2018-03-09:
What I would recommend is to either use a central location for your MIDI handling code (as I suppose your application is doing more than just MIDI stuff), and expose the higher level interface to the rest of your code. This way, you would be able to `#include <MIDI.h>` (after properly setting up the include paths) in your interface header, and hide the MIDI object away.

The macro acts both as a typedef and an object creation, you can split that in two to typedef in the header and create a local object in the implementation file.

nomadic68, 2018-03-25:
Hi franky47, I looked into the library source code and found that MIDI_CREATE_DEFAULT_INSTANCE();
corresponds to 
midi::MidiInterface<HardwareSerial> MIDI((HardwareSerial&)Serial);

I really need to separate my MIDI handling code in 2 files, so I am trying to figure out what is the corresponding extern declaration for the MIDI object in the 2nd source file, that is something like

extern template class MidiInterface MIDI;  // this does not work

I didn't find the typedef you were mentioning, unless it is the one related to Settings.

Can you please provide me the correct extern declaration for the MIDI object? This C++ is too advanced for me (by the way: congratulations for your awesome job!!!)

Thanks and regards.

nomadic68, 2018-03-25:
No worries i found it on my own (i must be getting old...), i share it to whom it may concern:

following a MIDI_CREATE_DEFAULT_INSTANCE();

the extern declaration for other source files is:

extern midi::MidiInterface<HardwareSerial> MIDI;

Cheers

closed 2018-03-25 by nomadic68

jmeile, 2019-11-28:
Hi nomadic68. Could you please post an example? I'm trying to accomplish this, but I don't understand where to put the "extern" declaration and the "MIDI_CREATE_DEFAULT_INSTANCE();" callback

jmeile, 2019-12-01:
Never mind. I figured out this. See my issue #135.