FortySevenEffects/arduino_midi_library #96: MidiUSB Only Works on Channel 1

neilmendoza, 2018-03-14:
Arduino only responds on channel 1 when using MidiUSB.

krizroring, 2018-07-14:
MIDI.begin() defaults to only listen to midi channel 1, to listen to all the channels use:  MIDI.begin(MIDI_CHANNEL_OMNI);

neilmendoza, 2018-07-18:
OK, thanks. The opposite behaviour seems more logical to me, MIDI.begin() listens to all channels and you pass a channel number to filter.

franky47, 2020-04-14:
After some consideration, I think listening to all channels by default is a specific use-case for some particular devices. Having a silent breaking change (all other devices suddenly receiving all channels) too seems dangerous a trade-off, so we'll keep the default to channel 1, and require an explicit opt-in to listening to all channels via `MIDI.begin(MIDI_CHANNEL_OMNI)`.

closed 2020-04-14 by franky47