FortySevenEffects/arduino_midi_library #102: Issues with Sending MIDI commands via Native USB port on an Arduino DUE

mlordin, 2018-10-10:
I am using version 4.3.1 of the MIDI Library. The Arduino IDE is 1.8.7 on Windows 10. I am using an Arduino DUE board and aiming to use the Native USB capacity of the board. I am also using the "UsbTransport" layer functionality within the library and can successfully receive MIDI commands via callback functions coming in from the outside - typically generated via MIDI-OX. However when I am sending MIDI commands from the Arduino Due via USB, e.g. Program Change, everything seems to work as it should but no events are recorded or displayed in MIDI-OX. I am also using two "traditional" 5-pin DIN ports via hardware serial ports Serial1 and Serial2 and they work fine in both directions using the library.

I am unfortunately not a savvy programmer and have tried to go through and trace parts of the MIDI Library code but can't figure out what is wrong.

// The MIDI ports are defined as follows
MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI_PORT_1);  // DIN Midi via DUE Serial1 pins
MIDI_CREATE_INSTANCE(HardwareSerial, Serial2, MIDI_PORT_2);  // DIN Midi via DUE Serial2 pins

static const unsigned sUsbTransportBufferSize = 64;
typedef midi::UsbTransport<sUsbTransportBufferSize> UsbTransport;
UsbTransport sUsbTransport;
MIDI_CREATE_INSTANCE(UsbTransport, sUsbTransport, MIDI_PORT_3);

// In the setup function - the callback functions for each port is declared
// Declare the MIDI callback functions for MIDI Port 1 - Serial1
MIDI_PORT_1.setHandleProgramChange(myHandleProgramChangePort1);
MIDI_PORT_1.setHandleControlChange(myHandleControlChangePort1);
MIDI_PORT_1.setHandleSystemExclusive(myHandleSystemExclusivePort1);
// Initiate MIDI communications, listen to specific channel from General MIDI Settings
MIDI_PORT_1.begin(myGenMIDISet.midiChannel1);

// Declare the MIDI callback functions for MIDI Port 2 - Serial2
MIDI_PORT_2.setHandleProgramChange(myHandleProgramChangePort2);
MIDI_PORT_2.setHandleControlChange(myHandleControlChangePort2);
MIDI_PORT_2.setHandleSystemExclusive(myHandleSystemExclusivePort2);
// Initiate MIDI communications, listen to specific channel from General MIDI Settings
MIDI_PORT_2.begin(myGenMIDISet.midiChannel2);

// Declare the MIDI callback functions for MIDI Port 3 - Native USB
MIDI_PORT_3.setHandleProgramChange(myHandleProgramChangePort3);
MIDI_PORT_3.setHandleControlChange(myHandleControlChangePort3);
MIDI_PORT_3.setHandleSystemExclusive(myHandleSystemExclusivePort3);
// Initiate MIDI communications, listen to specific channel from General MIDI Settings
MIDI_PORT_3.begin(myGenMIDISet.midiChannel3);

// The loop function  is pretty straight forward
MIDI_PORT_1.read();
MIDI_PORT_2.read();
MIDI_PORT_3.read();

// When pressing one of the hardware buttons connected to tmy Due board the following code is executed:
MIDI_PORT_1.sendProgramChange(data1, channel); // Works perfectly
MIDI_PORT_2.sendProgramChange(data1, channel); // Works perfectly
MIDI_PORT_3.sendProgramChange(data1, channel); // Nothing registered in MIDI-OX

As mentioned earlier - all MIDI communication over MIDI_PORT_1 and MIDI_PORT_2 works fine in both directions along with the callback functions for MIDI in over MIDI_PORT_3 (USB port) However, when sending e.g. Program Change from my program nothing is recorded/displayed at the receiving end e.g. MIDI-OX. 

I have tried varying the MIDI Default setting, and especially changing the Baud rate (115200 / 31250) without success:

struct myMIDI_USBSettings : public midi::DefaultSettings{
    static const bool UseRunningStatus = false;
    static const bool HandleNullVelocityNoteOnAsNoteOff = true;
    static const bool Use1ByteParsing = true;
    static const long BaudRate = MIDIDINBAUDRATE; // 31250;
    static const unsigned SysExMaxSize = MIDISYSEXMAXSIZE;
};
MIDI_CREATE_CUSTOM_INSTANCE(UsbTransport, sUsbTransport, MIDI_PORT_3, myMIDI_USBSettings);

As stated earlier, I have been struggling with this for hours and tried to Google for answers without success. Any ideas, tips or hints are most appreciated.

Regards,
Mats

mlordin, 2018-10-11:
For what it's worth, I have now also tried one of the simple MIDIUSB library examples where I have similar problems sending MIDI commands, albeit the behavior is a bit different - the first MIDI command is sent to MIDI-OX (MIDI monitor) and the nothing seems to happen despite the sketch looping and repeatedly sending MIDI commands until I stop the Serial Monitor and turn it back on - at that point all the "cached" MIDI events are quickly displayed within MIDI-OX prior to again stopping at the first command from the newly initiated sketch. The behavior is consistent and can be repeated. I posted a topic on the Arduino forum about this as well.
[https://forum.arduino.cc/index.php?topic=572968.0](url)

mlordin, 2018-10-16:
Update - the MIDI library seems to work fine but MIDI-OX is flaky as a monitoring tool when it comes to registering midi event via USB properly on Windows 10. After receiving some input from my posting on the Arduino Forum - see link below - I reverted to creating a MIDI track in Cubase with a MIDI Monitor plug-in. The MIDI events sent from my Arduino Due are now recorded properly within Cubase. Others testing the DUE setup have had success with e.g. Reaper. 

https://forum.arduino.cc/index.php?topic=572968.0

franky47, 2018-10-31:
I've found many issues in my code with the sending side of USB MIDI, as it was experimental I did not fully test it, so I'm going to write some unit tests and push a fix when possible.

franky47, 2018-11-06:
Yes, I've got a working fix on the [`feat/4.4.0`](https://github.com/FortySevenEffects/arduino_midi_library/tree/feat/4.4.0) branch, you'll also need to get the latest code from `master` on [`MIDIUSB`](https://github.com/arduino-libraries/MIDIUSB) until the next release of MIDIUSB is published.

mlordin, 2018-11-06:
Hi franky47 - followed your instructions and downloaded the updated MIDIUSB (master) library along with the MIDI Library (feat/4.4.0). I only had time to make some quick tests but there are definitely improvements when monitoring the MIDI events within MIDI-OX and sending commands via USB - MIDI-OX now displays the same MIDI messages immediately independently if they are sent via my serial 5-pin DIN ports or the USB port. I will do further in-depth tests later this week when I get a chance but appreciate your time and effort in providing the update which seems to be a huge improvement!

Cheers!
Mats

closed 2019-02-22 by franky47