FortySevenEffects/arduino_midi_library #103: Issues with Arduino Leonardo

sgarza, 2018-10-11:
Hi, I am using version 4.3.1 of the MIDI Library. The Arduino IDE is 1.8.7 on MacOS Sierra

I could not get to work the following code:

```
#include <MIDI.h>
#include <midi_UsbTransport.h>

static const unsigned sUsbTransportBufferSize = 16;
typedef midi::UsbTransport<sUsbTransportBufferSize> UsbTransport;

UsbTransport sUsbTransport;

MIDI_CREATE_INSTANCE(UsbTransport, sUsbTransport, MIDI);

const int velocity = 127; //Max Velocity (range is 0-127)
const int channel = 1; //MIDI Channel 1 (out of 16)

void OnControlChange(byte channel, byte control, byte value) {
  Serial.print("Control Change, ch=");
  Serial.print(channel, DEC);
  Serial.print(", control=");
  Serial.print(control, DEC);
  Serial.print(", value=");
  Serial.print(value, DEC);
  Serial.println();
  
}

void handleNoteOn(byte inChannel, byte inNumber, byte inVelocity)
{
    Serial.print("NoteOn  ");
    Serial.print(inNumber);
    Serial.print("\tvelocity: ");
    Serial.println(inVelocity);
}

void handleNoteOff(byte inChannel, byte inNumber, byte inVelocity)
{
    Serial.print("NoteOff  ");
    Serial.print(inNumber);
    Serial.print("\tvelocity: ");
    Serial.println(inVelocity);
}

void handleClock () {
  Serial.println('tick', HEX);
}

void setup() {
    Serial.begin(115200); 
    
    MIDI.setHandleControlChange(OnControlChange);
    MIDI.setHandleNoteOn(handleNoteOn);
    MIDI.setHandleNoteOff(handleNoteOff);
    MIDI.setHandleClock(handleClock);
    MIDI.begin();
    MIDI.sendProgramChange (18,1);
}

void loop() {
    if (MIDI.read()) {
      Serial.println(MIDI.getType(), HEX);
    }
    
   for (int note = 53; note <= 93; note ++) {
    MIDI.sendNoteOn(note, velocity, channel);  // Turn the note on.
    delay(200);                               // Wait 1000 milliseconds.
    MIDI.sendNoteOff(note, 0, channel); // Turn the note off.
  }
}
```

The handler `setHandleControlChange(OnControlChange)` works when I start/stop the transport in ableton.

But `Midi.sendNoteOn` doesn't get registered on Ableton midi input. It's weird cause it works if I use the `MIDIUSB` library directly.

This is the log of the midi monitor, only the `noteOff` are registered there [image]

But in ableton there is no midi input blink as seen here: https://monosnap.com/file/ggHYT99YFeDh08gRNxmuhNjS60NXqc

Cannot find anything related on the intertet. I've ran lots of examples without success :(

franky47, 2018-10-30:
Two things come in mind:
- Conflict between serial debugging and the USB MIDI functionnality (does Leonardo allow serial communications in parallel of MIDI endpoint enumeration ?)
- Your `for` loop inducing 8 seconds of delay (40 notes * 200ms each), during which the library is not reading anything.

franky47, 2018-10-31:
Ok so I found many issues with my code on the sending side of USB MIDI, I'll try and push a fix asap.

sgarza, 2018-11-01:
I'm not sure what this means `does Leonardo allow serial communications in parallel of MIDI endpoint enumeration ?` sorry =/

franky47, 2018-11-01:
Sorry I was thinking out loud, I was wondering if it was possible to use both MIDI and serial debugging through the same USB connection on a Leonardo. It is indeed possible, so your issue is most probably linked to those bugs I found in the USB interface code.

I'll let you know when I have a fix ready.

sgarza, 2018-11-01:
Awesome! thanks!! and yes the USB Serial debugging and USB MIDI can work at the same time on the Leonardo :)

franky47, 2018-11-06:
I've got a working fix on the [`feat/4.4.0`](https://github.com/FortySevenEffects/arduino_midi_library/tree/feat/4.4.0) branch, you'll also need to get the latest code from `master` on [`MIDIUSB`](https://github.com/arduino-libraries/MIDIUSB) until the next release of MIDIUSB is published.

sgarza, 2018-11-06:
Awesome thank you!!

sgarza, 2018-11-07:
Everything is working as expected, thank you again 👍

closed 2018-11-07 by sgarza