FortySevenEffects/arduino_midi_library #104: Time signature

Jimbolaya64, 2018-10-28:
Hello. I’m interested and in using this library to develop a midi controller to link my Beat Buddy drum machine to my guitar effects processor. Part of this requires being able to read a midi time signature. From my amateurish review of the code I don’t see this feature. Any plans to include it? The time signature is described here: 
[http://midi.teragonaudio.com/tech/midifile/time.htm](url)
Thank you.

Jimbolaya64, 2018-10-28:
Thanks for the quick reply. I’m trying to sync drum beats from the Beat Buddy to a Boss RC-20XL looper pedal. I want to start and stop the looper at the start of a measure. To determine the start of a measure I need the midi clock, a start signal and the time signature. The Beat Buddy puts all of these out. I can read the clock and start signal with your library. I don’t think I can read the time signature. Does that better describe what I’m trying to do?

Jimbolaya64, 2018-10-29:
This link describes it well:
[http://midi.teragonaudio.com/tech/midifile/time.htm](url)
I did check and this is exactly what the BB puts out for time signature.

mink99, 2018-10-29:
Once any device receives this message it will immediately shutdown as the signature 0xFF, if not used in a smf, is the system reset message.... before implementing that killswitch in the midi library I would strongly recommended to ask the developers of the beat buddy to change their code....

Jimbolaya64, 2018-10-29:
The BB developers and the developers of the midi capable looper pedals (Boomerang III, Pigtronix Infinity, TC Electronic Ditto X4, Boss RC-300) have been working together for a few years now and agreed on the use of the system reset message with a time signature in direct device communication. I doubt they will change anything at my request. The midi library already has the system reset message implemented. I just need that sub-routine to return the data for the time signature.

mink99, 2018-10-29:
Only because some developers have agreed to contradict the midi standard there should be a fixup in the midi library that will definitely harm any device that has not been implemented by these some developers ? Man you must be joking...... just create a switch on your device that will define the signature .

mink99, 2018-10-29:
I will as a colleague who owns a bb to file a bug report on this behaviour on the bb . The response will be published in the midi.org discussion forums.....

Jimbolaya64, 2018-10-29:
Sorry. I don't know the midi standard. That's why I asked the question. I learn a little more with each response that I get. No jokes. As far as a switch goes... I would prefer not to use one. I'm trying to automate/simplify things so as to minimize error and equipment failure during my live performance. Thank you for your guidance. I will move on.

mink99, 2018-10-29:
Sorry , not meant to be disrespectful.....

mink99, 2018-10-29:
Asked the colleague to provide a trace , his response was : 
_I switched Time Signature ON, started BB and switched to a 5/4 rhythm out of the OddTime-Folder. The BB sent this message:
F0 7F 7F 03 02 04 05 02 18 08 F7
I hooked it up to my MicroX again, did the same and it plays as usual. Note that the BB stops when you change songs/Rhythms._

So your basic assumption may have been wrong.....

Jimbolaya64, 2018-10-29:
I agree with what your colleague got from the BB. I did do some tracing. My error was assuming what I saw in a Google search was what I saw from the BB. I didn’t compare the data side by side. Thank you for following through. I believe this issue is resolved.

closed 2018-11-03 by franky47