FortySevenEffects/arduino_midi_library #114: Need help in detecting a bank change

chipmc, 2019-03-19:
I am just getting started working with MIDI and this library has been fantastic!

I have been able communicate with a Korg Trinity keyboard and have been able to send micro-tuning SysEx messages successfully.  Unfortunately, these SysEx messages change slightly based on the bank selected on the keyboard itself. I can see the commands coming from the keyboard - here are the messages from three bank changes:
0	To Port1	Control	1	B0 00 00
0	To Port1	Control	1	B0 20 **00**
0	To Port1	Program	1	C0 00
15:16:16.287	To Port1	Control	1	B0 00 00
15:16:16.288	To Port1	Control	1	B0 20 **01**
15:16:16.288	To Port1	Program	1	C0 00
15:16:18.384	To Port1	Control	1	B0 00 00
15:16:18.385	To Port1	Control	1	B0 20 **04**
15:16:18.385	To Port1	Program	1	C0 00

What I need to do is extract the one byte (bolded above) that will tell me what bank is selected which, in turn, will help me set the right SysEx messages.  

The problem is that I don't see a way of capturing these messages as they do not show up as program or control changes using getType().  

Any suggestions would be greatly appreciated.  

Thanks, Chip

franky47, 2019-03-20:
Could it be that they are sent on a different channel ? Do you see them if you set the input channel to Omni ?

chipmc, 2019-03-20:
@franky47 ,

Thank you for taking a look.  This message came in on channel 1 but I am listening to all channels using the command MIDI.begin(MIDI_CHANNEL_OMNI); 

Am I correct in assuming that these messages should be recognized as ContolChange using the getType() as I am attempting in my code now.  I am not using the handler  - MIDI.setHandleControlChange()  - would that make a difference?

Thank you again for your help.  Chip

franky47, 2019-03-20:
Both `getType` and the callback should work identically (the callback will be called first as it runs from within `MIDI.read()`).

Are you able to receive any other type of messages (NoteOn/Off for example) ? If not, can I have a look at your sketch file ?

chipmc, 2019-03-20:
@franky47 , 

I am able to receive and send all other sorts of messages, for example, this case receives a NoteOn or a NoteOff and sends it back out with the appropriate micro tuning.  It is only the continuous controller messages that are being missed.  

`     case midi::NoteOff:
        MIDI.sendNoteOff(MIDI.getData1()+sevenseg_value, 0, MIDI.getChannel());                 // Turn the note off when requested
        break;
     case midi::NoteOn:
        MIDI.sendNoteOn(MIDI.getData1()+sevenseg_value, MIDI.getData2(),MIDI.getChannel());     // Transpose based on value on 7-Segment display
        break;
`

I will send you an email with a link to the repository.  

Thanks, Chip

chipmc, 2019-03-20:
I have created a minimal sketch so you don't have to plow though the 800+ lines of code in my main file.  I will paste the sketch below and attach the MIDI test file that should give you the same inputs.

*** Could there be contention between my Serial debug messages and the MIDI on the same serial port? ***

`// Minimal code to troubleshoot CC issue
#include <MIDI.h>  // Add Midi Library

//Create an instance of the library with default name, serial port and settings
MIDI_CREATE_DEFAULT_INSTANCE();

void setup() {
  MIDI.begin(MIDI_CHANNEL_OMNI);                  // Initiate MIDI communications, listen to all channels
  MIDI.turnThruOff();                             // We will want to control what goes out
  MIDI.setHandleControlChange(MyCCFunction);      // This command tells the MIDI Library to call "MyCCFunction" when a continuous controller message is received
  Serial.println("Setup complete");
}

void loop() { // Main loop
  MIDI.read(); // Continuously check if Midi data has been received.
}

// Function will be passed Channel, Controller Number, and Value
// I am interested in the Bank Change messages 00 and 20.
// http://www.ccarh.org/courses/253/handout/controllers/controllers.html
void MyCCFunction(byte channel, byte number, byte value) {
  switch (number) {
    case 00:
      Serial.print("Bank Change coarse: ");
      Serial.println(value,HEX);
      break;
    case 20:
      Serial.print("Bank Change fine: ");
      Serial.println(value,HEX);
      break;
    default:
      Serial.println("Unknown Continuos Controller Message");
      break;
  }
}`

[Program Bank Change.midi.zip](https://github.com/FortySevenEffects/arduino_midi_library/files/2988657/Program.Bank.Change.midi.zip)

franky47, 2019-03-22:
Ah, yes I believe this could be a clash between the MIDI and Serial ports. What board are you on ?

chipmc, 2019-03-22:
I have a custom board where I have two serial ports - the standard one and a software serial one.  Still, even if I don't load the Software serial, I get this issue.  In the example above the MIDI library is the only one listening to the default Arduino serial port. 

To rule out the potential of conflict, I will create a new version where there is no SoftwareSerial.  Thank you again for your help and I will let you know how it goes.

Chip