FortySevenEffects/arduino_midi_library #116: When I add the library it doesnt compile

jwoodlee, 2019-04-21:
Arduino Uno ,  Ide 1.8.9 

Arduino/libraries/MIDI/midi_Defs.h:170:5: error: expected ',' or ';' before 'midi'
     midi::MidiInterface<Type> Name((Type&)SerialPort);

Arduino/libraries/MIDI/midi_Defs.h:182:9: note: in expansion of macro 'MIDI_CREATE_INSTANCE'
         MIDI_CREATE_INSTANCE(HardwareSerial, Serial,  MIDI);

Code/Arduino/buttonSwitch/buttonSwitch/buttonSwitch.ino:15:1: note: in expansion of macro 'MIDI_CREATE_DEFAULT_INSTANCE'
 MIDI_CREATE_DEFAULT_INSTANCE();
 ^
exit status 1
Error compiling for board Arduino/Genuino Uno.
         ^

franky47, 2019-04-22:
The error suggests you may have forgotten a semicolon on the line preceding the invokation of `MIDI_CREATE_DEFAULT_INSTANCE`.

jwoodlee, 2019-04-22:
But it’s in your midi library 

Jason
C 862-400-8936

jwoodlee, 2019-04-22:
this is the area having an issues in midi_Defs.h

/*! \brief Create an instance of the library attached to a serial port.
 You can use HardwareSerial or SoftwareSerial for the serial port.
 Example: MIDI_CREATE_INSTANCE(HardwareSerial, Serial2, midi2);
 Then call midi2.begin(), midi2.read() etc..
 */
#define MIDI_CREATE_INSTANCE(Type, SerialPort, Name)                            \
    midi::MidiInterface<Type> Name((Type&)SerialPort);

#if defined(ARDUINO_SAM_DUE) || defined(USBCON)
    // Leonardo, Due and other USB boards use Serial1 by default.
    #define MIDI_CREATE_DEFAULT_INSTANCE()                                      \
        MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI);
#else
    /*! \brief Create an instance of the library with default name, serial port
    and settings, for compatibility with sketches written with pre-v4.2 MIDI Lib,
    or if you don't bother using custom names, serial port or settings.
    */
    #define MIDI_CREATE_DEFAULT_INSTANCE()                                      \
        MIDI_CREATE_INSTANCE(HardwareSerial, Serial,  MIDI);
#endif

/*! \brief Create an instance of the library attached to a serial port with
 custom settings.
 @see DefaultSettings
 @see MIDI_CREATE_INSTANCE
 */
#define MIDI_CREATE_CUSTOM_INSTANCE(Type, SerialPort, Name, Settings)           \
    midi::MidiInterface<Type, Settings> Name((Type&)SerialPort);

END_MIDI_NAMESPACE

franky47, 2019-04-22:
Can I please have a look at your sketch file ?

jwoodlee, 2019-04-22:
Oddly now its compiling after a restart. Close the issue.

closed 2019-04-22 by jwoodlee