FortySevenEffects/arduino_midi_library #118: sendProgramChange sends incremented program number

greizdeifi, 2019-06-09:
Hi,

if I send a program change message, e.g. something simple as
`#include <MIDI.h>

MIDI_CREATE_DEFAULT_INSTANCE();

bool first = true;

void setup()
{
    MIDI.begin(1);                      // Launch MIDI and listen to channel 1
}

void loop()
{
    if (first == true) {
      MIDI.sendProgramChange(102, 1);
      first = false;
    }
}`

then instead of inted value 102 the incemented value 103 is received. If I do e.g. a control change message with the same value theresult is correct and as expected. 
For me, all program numbers always get incremented by 1. Can someone confirm this behavior?

Thanks.

mink99, 2019-06-09:
Many devices show program changes as 1-128 while receiving 0-127.

franky47, 2019-06-09:
@mink99 is correct, the library, being a low level interface, uses zero-indexed values, apart from channel numbers which are usually reasoned about as one-indexed:
http://www.somascape.org/midi/tech/spec.html#statCn

closed 2019-06-09 by franky47