FortySevenEffects/arduino_midi_library #121: question : How to directly send bytes over MIDI interface?

binooetomo, 2019-07-20:
Dear all.

Currently i'm trying to write kind of USB_MIDI-SERIAL_MIDI bridge on a ProMicro
for the USB_MIDI part, I use USBMIDI library

1. messages coming via serial_midi will forwarded via USB_MIDI ... it's easy since currently i'll only handle NoteOn and NoteOff comming via serial side.

2. But messages comming via USBMIDI will not just that two types.
example of USBMIDI at https://github.com/arduino-libraries/MIDIUSB/blob/b93e9fe3b65750d120497f735f595bb1a7cf8bd7/examples/MIDIUSB_read/MIDIUSB_read.ino only show that the 'read' will only give me : header plus 3 bytes.
My question is, how to 'just send' that 3 byte using MIDI.h ?

Sincerely
-bino-

franky47, 2019-07-20:
While there is already a [USB transport](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/MidiUSB/MidiUSB.ino), it lacks the ability to directly transcribe an arbitrary message from USB to serial, but this is intentional: USB MIDI packets can carry only part of a message (ex: SysEx being split across multiple packets), and therefore the transport does not mirror the incoming data directly to the serial output to avoid collisions with messages sent from within the running sketch.

I would recommend to use callbacks to handle the messages you expect to receive from USB and send them back with the serial interface, using something like this:

```cpp
#include <MIDI.h>

#if defined(USBCON)
#include <midi_UsbTransport.h>

static const unsigned sUsbTransportBufferSize = 16;
typedef midi::UsbTransport<sUsbTransportBufferSize> UsbTransport;

UsbTransport sUsbTransport;

MIDI_CREATE_INSTANCE(UsbTransport, sUsbTransport, UsbMidi);
MIDI_CREATE_INSTANCE(HardwareSerial, Serial, SerialMidi); // May need to be changed to the default hardware serial port

#else
#error "No USB available"
#endif

// --

void handleUsbNoteOn(byte channel, byte number, byte velocity)
{
    SerialMidi.sendNoteOn(number, velocity, channel);
}

void handleUsbNoteOff(byte channel, byte number, byte velocity)
{
    SerialMidi.sendNoteOff(number, velocity, channel);
}

void setup()
{
    UsbMidi.begin();
    UsbMidi.setHandleNoteOn(handleUsbNoteOn);
    UsbMidi.setHandleNoteOff(handleUsbNoteOff);
    SerialMidi.begin();
}

void loop()
{
    UsbMidi.read();
    SerialMidi.read();
}
```

Now, to actually answer your question, if you want to send raw bytes out on the MIDI serial port, it's just a matter of using the [same serial port](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/src/midi_Defs.h#L217) as the MIDI interface does:
```cpp
Serial.write(0x42);
```

binooetomo, 2019-07-22:
Dear Sir.
I really appreciate your response.

I'll try your first recommendation.

closed 2019-07-22 by binooetomo