FortySevenEffects/arduino_midi_library #123: Allow for other MIDI transports like ipMIDI, AppleMIID, BLE MIDI

lathoub, 2019-08-09:
related to closed issue #74 

Topic: making this library open for other transport protocols (ipMIDI, BLE MIDI, AppleMIDI, USB...)

The library actually already allow for an additional transport, over USB. The `read / write / available` maps nicely on the USB protocol. However AppleMIDI, BLE MIDI and ipMIDI or packet based and require an additional `beginPacket `and `endPacket`.

I have made a suggestion in [my fork of the library](https://github.com/lathoub/arduino_midi_library) extracting [all serial calls in a super class](https://github.com/lathoub/arduino_midi_library/blob/master/src/midi_Serial.h) and making MidiInterface abstract (renaming it AbstractMidiInterface) and adding :

```
	virtual void __beginWrite() {};
	virtual void __write(byte) = 0;
	virtual void __endWrite() {};
	virtual int  __available() = 0;
	virtual byte __read() = 0;
```

Is there a willingness to support this?
I have ipMIDI, AppleMIDI and BLE MIDI ready to support this.

Lathoub

lathoub, 2019-08-09:
I'm taking a closer look at midi_`UsbPacketInterface ` and it recomposes the incoming buffer. When a full messages is detected, then the packet can be send out, with a begin, send and endPacket. This does however make the code bigger that it needs to be and uses precious CPU cycles.

Do you want to hold on to this method?

franky47, 2019-08-09:
Hey, thanks for the interest, I'm on holiday away from a computer at the moment, I'll have a look in September.

Ideally, I'd like to avoid inheritance, I've used templates so far as they have no runtime cost.

lathoub, 2019-08-09:
Enjoy your vacation!

Here is food for thought for when you come back in September:
If you add `mSerial.flush(); ` at the end of each send block, you can use flush() as a delimiter for blocks in other that Serial transport protocols (for USB, no more need to call the package recomposer).

For Example:
```
…

        // Then send data
        mSerial.write(inData1);
        if (inType != ProgramChange && inType != AfterTouchChannel)
        {
            mSerial.write(inData2);
        }

// ADDED
        mSerial.flush();
...
```

With this comment: a call to [flush(); on HardwareSerial will have a decreased performance](https://www.baldengineer.com/when-do-you-use-the-arduinos-to-use-serial-flush.html), so needs to be avoided.
So similar to UsbTransport, I created a SerialTransport, that maps all serial calls 1-1, except for flush();

```
template<class SerialPort>
class SerialTransport
{
public:
    inline SerialTransport(SerialPort& inSerial)
    : mSerial(inSerial) {}
    inline ~SerialTransport() {}

public:
    inline void begin(unsigned inBaudrate)
    {
#if defined(AVR_CAKE)
        mSerial. template open<inBaudrate>();
#else
        mSerial.begin(inBaudrate);
#endif
    }

    inline unsigned available() { return mSerial.available();  }
    inline byte read() { return mSerial.read(); }
    inline void write(byte inData) { mSerial.write(inData); }
    inline void flush() {} // avoid that flush() is called on HardwareSerial, that has a big performance penanlty

private:
    SerialPort& mSerial;
};
```

The constructor in the example would then look like this (for a Mega):

```
#include <midi_SerialTransport.h>
typedef midi::SerialTransport<HardwareSerial> SerialTransport;
SerialTransport sSerialTransport((HardwareSerial&) Serial1);
MIDI_CREATE_INSTANCE(SerialTransport, sSerialTransport, MIDI);
```

I look forward to your comments in September!

lathoub, 2019-08-10:
A good night sleep brings good ideas... (?)

There is no need to name `flush()` the same as the Serial method when using templates; calling it `endPacket();` is much clearer. Having a end demarcation for the packet (endPacket()) is great but still needs an intermediate buffer to store all the bytes to be send. So creating a `beginPacket();` avoids that intermediate buffer.

In `midi.hpp`:

```
. . . 
template<class SerialPort, class Settings>
void MidiInterface<SerialPort, Settings>::sendTuneRequest()
{
    mSerial.beginPacket();
    mSerial.write(TuneRequest);
    mSerial.endPacket();

    if (Settings::UseRunningStatus)
    {
        mRunningStatus_TX = InvalidType;
    }
}
. . .
```

and in `midi_SerialTransport.h` :

```
template<class SerialPort>
class SerialTransport
{
public:
    inline SerialTransport(SerialPort& inSerial)
    : mSerial(inSerial) {}
    inline ~SerialTransport() {}

public:
    inline void begin(unsigned inBaudrate)
    {
#if defined(AVR_CAKE)
        mSerial. template open<inBaudrate>();
#else
        mSerial.begin(inBaudrate);
#endif
    }

    inline unsigned available() { return mSerial.available();  }
    inline byte read() { return mSerial.read(); }
    inline void beginPacket() {}
    inline void write(byte inData) { mSerial.write(inData); }
    inline void endPacket() {} 

private:
    SerialPort& mSerial;
};
```

lathoub, 2019-08-10:
Here is ipMIDI using templates (with begin and endPacket();)

```
#pragma once

#include "midi_Defs.h"

BEGIN_MIDI_NAMESPACE

template<class UdpClass>
class ipMidiTransport
{
public:
	inline ipMidiTransport()
		: mPacketBufferIndex(0), mPacketAmountRead(0)
	{
		mMultiIP = IPAddress(225, 0, 0, 37);
	};
	inline ~ipMidiTransport() {}

	inline bool begin(const int inPort = 21928, const Channel inChannel = 1)
	{
		mPort = inPort;
		return (mUdp.beginMulticast(mMultiIP, mPort));
	}

public :
	inline void beginPacket() { mUdp.beginPacket(mMultiIP, mPort); };

	inline void write(byte byte) { mUdp.write(byte); };

	inline void endPacket() { mUdp.endPacket(); };

	inline byte read() { return mPacketBuffer[mPacketAmountRead - mPacketBufferIndex]; };

	inline int available()
	{
		if (mPacketBufferIndex > 0)
			mPacketBufferIndex--; // data is still in the mPacketBuffer, empty it first
		else
		{
			auto packetSize = mUdp.parsePacket();
			if (0 == packetSize) return 0; // if nothing is available, leave here

			// data is ready to be read, do not read more than what we have memory for
			mPacketBufferIndex = mPacketAmountRead = mUdp.read(mPacketBuffer, sizeof(mPacketBuffer));
		}

		return mPacketBufferIndex;
	};

private:
	UdpClass mUdp;

	uint16_t mPort;
	IPAddress mMultiIP;

	byte		mPacketBuffer[UDP_TX_PACKET_MAX_SIZE];
	uint16_t	mPacketBufferIndex;
	uint16_t	mPacketAmountRead;
};

END_MIDI_NAMESPACE
```

lathoub, 2019-08-18:
I have re-forked repo branch feat/4.4.0, and made 2 branches: 1 that implements [dynamic polymorphism](https://github.com/lathoub/arduino_midi_library/tree/dynamic-polymorphism) and another using [static polymorphism](https://github.com/lathoub/arduino_midi_library/tree/static-polymorphism) - both extract the _transport_ mechanism out of the MidiInterface call, to allow for other transport mechanisms (USB, AppleMIDI, BLE, ipMIDI)

The branch using dynamic polymorphism is a easier to read, but has the overhead of table lookup. The branch with using static polymorphism is not as easy to read, but provides smaller code and provides better compile time checks. 

Compiling against the Leonardo on IDE 1.8.9:
`feat/4.4.0`: `7406 bytes` of program storage space and `511 bytes` of dynamic memory
`dynamic polymorphism`: `7278 bytes` of program storage space and `545 bytes` of dynamic memory
`static polymorphism`: `7258 bytes` of program storage space and `511 bytes` of dynamic memory

Compiling against the Leonardo on IDE 1.8.9 with #125:
`feat/4.4.0`: `7174 bytes` of program storage space and `510 bytes` of dynamic memory
`dynamic polymorphism`: `7024 bytes` of program storage space and `544 bytes` of dynamic memory
`static polymorphism`: `7024 bytes` of program storage space and `510 bytes` of dynamic memory

Interesting enough, the original code is larger the dynamic or static polymorphic branches.

lathoub, 2019-10-20:
Checking in with @franky47 . I have a working AppleMIDI stack using static polymorphism (ipMIDI also works on ESP32) on IDE 1.8.10. Are you considering making your library transport agnostic using static polymorphism? (makes the USB implementation a lot easier **and faster** as well - no need to reparse the MIDI messages). Merci!

franky47, 2019-10-21:
Hey, sorry I haven't had time to look at this, I'm in the process of moving houses and free time for open-source stuff is scarce, hopefully it should get better in the upcoming months.

lathoub, 2019-12-26:
Merry Christmas @franky47! I hope the 🏠 move went OK. Any chance you can have look at it? Have a great 2020 🥳 !

lathoub, 2020-03-25:
this above requests will be rolled up in release 5.0.0

closed 2020-03-25 by lathoub