FortySevenEffects/arduino_midi_library #128: Teensy 3.2 and Midi, recent issues

dimitre, 2019-09-23:
Hello, I've been using this library with Teensy 3.2. I'm using the default TX pin and it always worked ok with MIDI.begin()
in the latest version now it is sending all MIDI information to the serial port instead of TX pin. not sure if it is this library or something else.
Happening both on Arduino 1.8.9 and 1.8.10
minimal code to reproduce
```c++
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();

void setup() {
  MIDI.begin();
}

void loop() {
  MIDI.sendNoteOn(42, 127, 1);    // Send a Note (pitch 42, velo 127 on channel 1)
  delay(1000);                // Wait for a second
  MIDI.sendNoteOff(42, 0, 1);     // Stop the note
  delay(100);
  // put your main code here, to run repeatedly:

}
```

[image]

Thank you

dimitre, 2019-09-23:
It seems to be working on latest master
But I have this issue if I install via Arduino Libraries

hurzl, 2021-09-17:
Is this expected? I have the same on mega1280. Same for master and arduino installed versions.
My MIDI device which is connected via TX>1 detects random nonsense commands.