FortySevenEffects/arduino_midi_library #130: Does sendAfterTouch change the Velocity of a note being played?

samlochner, 2019-10-23:
Hi, I'm trying to build a digital woodwind instrument. I'm new to Arduino, midi, and this library, but I've got a very basic example working where I press a button and the Arduino sends a note through a midi to usb cable to the iPhone app GarageBand. I've also got a pressure sensor to control the velocity of the note. This creates a nice steady note. But I want the volume to change as the pressure changes.

If I understand correctly, I can use sendAfterTouch to change the velocity of the note after it has already been started (is this correct?). I've tried doing this but it doesn't seem to be working in Garageband iOS. Does anyone know why not and how I can make it work? Or perhaps you know of another app that could do this?

Thanks!
Sam

franky47, 2019-10-23:
AfterTouch does not change velocity, how it's interpreted is up to the receiving instrument. You could probably assign it to volume change on some instruments, but I think you'd have better luck sending ControlChange number 7, Channel Volume, to do what you're looking for.

A list of all control names can be found here:
http://www.somascape.org/midi/tech/spec.html#ctrlnums

However, like AfterTouch, it's always up to the receiving instrument to define what to do from each control.

samlochner, 2019-10-24:
Thanks you for the detailed response franky47. ControlChange 7 did the trick for me. I came across another app called Sensual Sax that shows this mapping too, plus some other interesting ones I'll have to try out. I'm hoping to find more apps like this for other instruments.

closed 2019-10-24 by samlochner