FortySevenEffects/arduino_midi_library #131: Work directly over USB?

samlochner, 2019-10-24:
I've been able to get this working by using a midi to USB cable. But is it possible to go directly from the USB port on the Arduino then to the computer or better yet Iphone (via camera adapter)? I'm using a Leonardo, so that should help.

Thanks,
Sam

franky47, 2019-10-24:
It's possible with some Arduinos, like the Leonardo, which has a built-in USB device interface.

You can have a look at the USB examples, using MIDIUSB.

samlochner, 2019-10-24:
I did try the MIDIUSB examples and directly over USB it worked when hooked up to a computer, but not on the iPhone (via camera adapter and powered usb hub). I asked over at the [MIDUSB issues](https://github.com/arduino-libraries/MIDIUSB/issues/58) but haven't gotten a response yet.

I was thinking this arduino midi library would have its own way, but perhaps not...

brorgustav, 2019-10-24:
For me it has worked With midiusb library to iPhone. When using a micro, but it should be no difference, same chip.

samlochner, 2019-10-24:
That's promising to hear, brorgustav. Did you have to use a powered usb hub? I find if I go from the arduino usb to the camera adapter I get a power error on the iphone, so I put a usb hub inbtween. Perhaps that has something to do with it?

franky47, 2020-04-20:
Version 5.0.0 has been released, which brings a new transport system for USB MIDI. Check out the README for more information.

closed 2020-04-20 by franky47