FortySevenEffects/arduino_midi_library #132: Working on Leonardo but Not Due

samlochner, 2019-10-29:
Hi, I've got the Basic_IO example working on an Arduino Leonardo but for some reason it won't work on a Leonardo Due for me.

Curiously, the basic Midi example (does not require arduino_midi_library) works fine for me on both Leonardo and Due.

Anyone have any idea why the Due is having trouble running the Basic_IO example?

Thanks,
Sam

samlochner, 2019-10-30:
Figured it out.

Instead of using: MIDI_CREATE_DEFAULT_INSTANCE();
I use:

struct MySettings : public midi::DefaultSettings
{
    static const unsigned SysExMaxSize = 1024; // Accept SysEx messages up to 1024 bytes long.
    static const bool UseRunningStatus = true; // My devices seem to be ok with it.
};

// Create a 'MIDI' object using MySettings bound to Serial.
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);

samlochner, 2019-10-30:
I seem to be getting somewhere by creating a custom instance instead of default and using Serial instead of Serial2.
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI, MySettings);

However, it is very flaky, and when I upload to the board while connected to iOS garageband, I get all these interesting notes and all the settings are messed with. It seems somehow a bunch of stuff is being sent over the Midi cable that isn't supposed to be...

Hoping someone can help me out here...

Thanks!
Sam

dblscore, 2025-02-15:
I also had some problems with Arduino Due, but the problem comes from the serial port used. 
By default Serial1 must be connected to pins RX1 and TX1 which are not in the same position as the RX and TX of the Leonardo. 
So by changing the pins in my case everything worked.