FortySevenEffects/arduino_midi_library #136: Any pure C++ version of this?

jmeile, 2019-12-06:
Hi

I'm just wondering if there is any C++ pure version of this library. I would like to create a MIDI project, but on a Raspberry Pi. The only thing I have found so far is rtmidi; however, it only allows to define a callback for raw MIDI messages, so, I will have to parse the MIDI message to know if it is a NOTE ON, a CONTROL CHANGE, etc..

I would like to have something as what you use here: different callbacks for the different MIDI messages: NOTE ON, NOTE OFF, CONTROL CHANGE, etc..

Thanks in advanced

Best regards
Josef

jmeile, 2019-12-07:
I changed the title and the description since the code is C++. I expressed it wrongly. I wanted to know if you can use it in other systems, ie: an Raspbery Pi or even a computer with linux and an USB to MIDI cable.

franky47, 2019-12-07:
The library uses templates to abstract the parsing from the transport used, so as long as you provide it with an interface that conforms to Arduino's Serial, it will work.

Have a look at this repository to see how to write an adapter for another platform:
https://github.com/FortySevenEffects/mbed-midi-adapter

closed 2019-12-07 by franky47