FortySevenEffects/arduino_midi_library #137: State of v4.4.0?

infovore, 2019-12-09:
Hello!

I've been poking this library for use with SAMD21/Arduino boards that will be running both USB and hardware-serial MIDI. It's a really nice set of abstractions.

However, 4.4.0 - currently only a branch - is very much necessary for working with USB-based MIDI. That branch has been open a _long_ time.

Is there any likelihood it'll get merged into master, which would help it show up in the Arduino IDE? What are the things that would help that happen?

mgamache, 2020-01-16:
I would like to use this lib with a Trinket M0 as a USB Host.

lathoub, 2020-03-04:
I just submitted a pull-request for `feat/4.4.0` lifting the Serial code from the standard library (using static polymorphism) into `SerialMIDI.h`. This allows for other serialisers, such as USBMIDI (but also [AppleMIDI](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/feat/2.0.0), ipMIDI, BLE-MIDI).

[AppleMIDI](https://github.com/lathoub/Arduino-AppleMIDI-Library/tree/feat/2.0.0) fully uses this mechanism (static polymorphism for serialisation) and works fine!

@franky47 has not been responsive, not sure if he has the time to continue to maintain this library

franky47, 2020-03-04:
Hey, I was about to write a response to @lathoub on #138, indeed I'm in lack of availability to maintain this library on my own, and therefore am open to accepting maintainers.

@lathoub, I appreciate your work on the last few issues and PRs, would you be interested ?

lathoub, 2020-03-04:
Hi @franky47 - thanks for the quick response :-)
Yes, interested to be co-maintainer (thank you for asking).

franky47, 2020-03-04:
OK, I have added you, if you want to chat off GitHub: contact@francoisbest.com

lathoub, 2020-03-04:
Cheers, I'll reach out

closed 2020-04-20 by franky47