FortySevenEffects/arduino_midi_library #139: Callbacks are not being fired

g105b, 2020-02-13:
Hi. I'm sure I'm doing something wrong, as I am trying to make use of such basic functionality and it is not working, but I have been working on this for such a long time I would like to ask for your advice.

Firstly, I have tested my instrument is sending correct MIDI by checking on my computer. I used Ableton and Web MIDI API to perform tests.

I'm using Arduino UNO for testing purposes. MIDI out cable of instrument is connected to Uno's RX pin and +5V pin. I have checked that they are correctly wired, even swapping the wires just in case.

I'm using version 4.3.1 of this library.

My code is simple: when the clock or a note comes through, turn on the builtin LED. There is no LED being lit, so I can assume the callback isn't being called.

For testing, I have made a more complex program that sends notes to the instrument, which works fine so I know the library has loaded correctly.

Simple program:

```c
#include <MIDI.h>

MIDI_CREATE_DEFAULT_INSTANCE();

void handleClock() {
  digitalWrite(LED_BUILTIN, HIGH);
}

void handleNoteOn(byte channel, byte note, byte velocity) {
  digitalWrite(LED_BUILTIN, HIGH);
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  MIDI.begin();
  MIDI.setHandleNoteOn(handleNoteOn);
  MIDI.setHandleClock(handleClock);
}

void loop() {
  MIDI.read();
}
```

Sorry if I'm being a total newbie, but I can't figure out what I'm doing wrong and I would love to continue using your library. Thanks!

franky47, 2020-02-13:
Usually, MIDI inputs require an optocoupler to work:
https://www.youtube.com/watch?v=GxfHijjn0ZM&list=PL4_gPbvyebyH2xfPXePHtx8gK5zPBrVkg&index=4&t=0s

I have managed to make it work without an optocoupler in some cases, but some devices (that respect the current loop specification) might be picky about this.

g105b, 2020-02-13:
Thanks for your fast reply!

Oh OK, I thought I would be fine to use my circuit before the optocouplers arrive from China. I'll park this to one side until then, as I'm sure there are no bugs with the library affecting me.

Thanks again for your work on the library.

closed 2020-02-13 by g105b

g105b, 2020-02-15:
Hey, just following up on this issue. I hooked up my midi device through an optocoupler and your library instantly saw the commands coming through, so it was as simple as that!

Thank you!

franky47, 2020-02-16:
I'm glad you got it working, thanks for the follow-up !