FortySevenEffects/arduino_midi_library #142: Roadmap to v5

franky47, 2020-03-11:
This issue tracks the scope, discussion and issue tracking for v5.

## Scope

v5 will include new transports:
- Full-featured USB MIDI
- Bluetooth BLE
- AppleMIDI

Transports are what convert raw data into MIDI messages (and vice versa). The default transport will remain Serial, for immediate DIN connections with RX/TX pins.

This should not introduce breaking changes in sketches, unless you use the `MidiInterface` definition directly. Macros will still work as before, and new ones will help with using the new transports. But because it introduces external libraries for those new transports and involves modifying a lot of the core code to handle packet-based transports (rather than stream-based like Serial), we're making it a major release.

Thanks a lot to @lathoub for his hard work on this ! 🥇

lathoub, 2020-03-11:
- ipMIDI is also available, but needs additional help (sending multicast UDP messages becomes unstable after a while). Appreciate all the help!

franky47, 2020-04-20:
5.0.0 has been released, thanks again @lathoub for your great work !

closed 2020-04-20 by franky47