FortySevenEffects/arduino_midi_library #143: Bank & program change message line 6 load setlist

pandeagus, 2020-04-07:
Hello.. i need help for load set list bank on 

Pod hd500 using midi
[image]
But i dont know the  code .. combine bank change cc0 (MSB) with cc32(LSB) and program change together... Please help me..

franky47, 2020-04-08:
This should do:
```cpp
MIDI.sendControlChange(0, 0, line6Channel);
MIDI.sendControlChange(32, setListNumber - 1, line6Channel);
MIDI.sendProgramChange(preset, line6Channel);
```

Replace:
- `line6Channel` with the MIDI channel of your Line6 POD (1 to 16)
- `setListNumber` with the Setlist to choose (1 to whatever maximum your POD supports)
- `preset` with the preset number (0 to 63 as described in the documentation)

pandeagus, 2020-04-08:
Not working... I dont why..
If working result like this

[image]
Use that code just preset change because program change.. 
Maybe you have another method for combine... Bank change cc0(bank msb) and cc32 (lsb)..
Sorry distrub your time

franky47, 2020-04-08:
Can I see your full code (or at least, the relevant part) ?

pandeagus, 2020-04-08:
#include <MIDI.h>
MIDI_CREATE_DEFAULT_INSTANCE();
int BankUp = 7;
int PresetUp = 8;
int pushed = 0;// 1 ditekan
int statusFS = 0;
int statusFS1 = 0;
int patch ;
void setup() {
  Serial.begin(9600);
  MIDI.begin();
  pinMode(BankUp,INPUT_PULLUP);
  pinMode(PresetUp,INPUT_PULLUP);
 }
void loop() {
 statusFS = digitalRead(BankUp);
 if(statusFS == LOW && pushed == 0){
      MIDI.sendControlChange(0,0,1);
      MIDI.sendControlChange(32,1,1);
      MIDI.sendProgramChange(1,1);
      delay(200);
 }  
 statusFS1 = digitalRead(PresetUp);
  if(statusFS1 == LOW && pushed == 0){
       patch++;
        MIDI.sendProgramChange(patch,1);
        delay(200);
  }
}

this is my code..

franky47, 2020-04-08:
Ok, what do you want it to do ? From what I understand, you have two footswitches (BankUp and PresetUp), what should happen when each is pressed ?