FortySevenEffects/arduino_midi_library #144: Retrieve port information over USB

Ordinateur-Hack, 2020-04-10:
Access MIDI Ports
---
I'm trying to connect my _ESP8266 Node MCU_ to my keyboard _Yamaha Tyros 3_. The latter offers two different MIDI ports ([reference manual, p. 116](https://de.yamaha.com/files/download/other_assets/9/314419/tyros3_en_rm_v112a.pdf)). I didn't find anything about connecting to specific MIDI ports neither in this GitHub repo, nor the [Arduino MIDI Library Documentation](http://fortyseveneffects.github.io/arduino_midi_library).
I try to do something as described [here](https://www.music.mcgill.ca/~gary/rtmidi/) in the RtMidiTutorial under the header _Probing Ports / Devices_ where they retrieve a list of available ports and their respective names. The [MidiUSB example](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/MidiUSB/MidiUSB.ino#L11) didn't work for me. I try to light up the inbuilt LED whenever the NodeMCU receives a Note_ON message from the keyboard, but it's not working. And I don't really know how I can debug this when I use the micro USB to connect to the keyboard...

Documentation for `MIDI_CREATE_INSTANCE()`?
---
Is there any documentation on `MIDI_CREATE_INSTANCE()`? In the examples, **I've seen that you are supposed to pass HardwareSerial and SerialUSB. How do they relate to the MIDI ports?** SerialUSB is something like "COM5", right?

```c++
static const unsigned sUsbTransportBufferSize = 16;
typedef midi::UsbTransport<sUsbTransportBufferSize> UsbTransport;

UsbTransport sUsbTransport;

MIDI_CREATE_INSTANCE(UsbTransport, sUsbTransport, MIDI);
```

I also don't quite get what [these lines](https://github.com/FortySevenEffects/arduino_midi_library/blob/master/examples/MidiUSB/MidiUSB.ino#L6-L11) are doing. For me - as someone from the Java and Node.js/Typescript world - it looks like you are declaring a new type named UsbTransport (with a specific buffer size that you pass in as generics) and then you declare a variable named sUsbTransport of that type but don't initialize it. How can these two things then be a hardware serial number and a serial USB number? (As you may have guessed, I'm relatively new to the C/C++-world.)

franky47, 2020-04-14:
Hi, we're planning on releasing v5.0.0 soon, which will change the way the library works with native USB transport.

As for the C++ part, we're indeed defining a new type, but the only variable there is the buffer size, the USB port and the identifier for the device were not configurable in that version.

@lathoub, is there a way to choose which USB port to use with your transport ?

lathoub, 2020-04-15:
The USB port selection is done by the [USBMIDI ](https://github.com/arduino-libraries/MIDIUSB)library, that does most of the magic behind the scenes. Not sure for the ESP8266, I can have a look

Ordinateur-Hack, 2020-04-15:
@franky47 @lathoub Ok great, thank you for your quick replys. I'll take a look into [USBMIDI ](https://github.com/arduino-libraries/MIDIUSB) as well and I'm looking forward to the new version v5.0.0 ;)

franky47, 2020-04-26:
Version 5 has been released, check out the README for information on how to get started with USB MIDI 🎉

closed 2020-04-26 by franky47

Ordinateur-Hack, 2020-04-28:
Wow, that's great, I'll try it out. Thank you.

Ordinateur-Hack, 2020-04-28:
@lathoub @franky47 Ok, I guess the NodeMCU v1.0 (ESP8266) board is not compatible with the underlying USBMIDI-library since I get the same error messages as described [here](https://github.com/arduino-libraries/MIDIUSB/issues/33).
Sadly, I won't be able to use your library. Do you know any alternatives for the NodeMCU?

franky47, 2020-04-28:
From what I can see in the datasheet, it's using a separate chip to translate UART to USB, so you might be able to pull it off by using [HairlessMIDI](http://projectgus.github.io/hairless-midiserial/) on your computer, a Serial interface on the NodeMCU, which talk to the CP2102 USB chip.

The complete chain would then be:
```
Signal flowchart:
Your code -> Serial MIDI -> NodeMCU TX -> CP2102 -> USB -> HairlessMIDI -> computer MIDI

Hardware connections:
[ESP8266] <- UART -> [CP2102] <- USB -> [Computer]
[ -------- NodeMCU -------- ]
```