FortySevenEffects/arduino_midi_library #145: MIDI.sendControlChange is not working

Imbr2, 2020-04-18:
Hi, I need help with this issue:
I'm using an Arduino Mega 2560 ATmega16U2 and the MIDI library to commuicate wiyh the computer via USB.
The will of my program is that when I press one button, the corresponding note is sent to the computer. Then, I use the command "MIDI.sendControlChange(volumeCommand, value, channel)" to adjust the volume of the previous note using a potentiometer. When I check my program in a synthesizer in the computer, the corresponding notes to each button sound properly, but the volume is never adjusted. The "sendControlChange" command is not working in my project.
Here I paste you the code:

`#include <MIDI.h>

const int analogSensor = A0;
int buttonApin = 9;
int buttonBpin = 8;
int sensorVal = 0;
int i = 0;
int p = 0;
MIDI_CREATE_DEFAULT_INSTANCE();

void setup()
{
  pinMode(buttonApin, INPUT_PULLUP);  
  pinMode(buttonBpin, INPUT_PULLUP);  
  MIDI.begin();
}

void loop()
{
  if (digitalRead(buttonApin) == LOW && i !=1) {
    sensorVal = (analogRead(analogSensor)/8);
    MIDI.sendNoteOn( 48, 127, 1);              // If "button1" is pressed, the note C is sended via channel 1.
    MIDI.sendControlChange(7, sensorVal, 1);   // Then, the volume (command 7) is adjusted to the "sensorVal" value.
    delay (2000);
    MIDI.sendNoteOff(48, 0, 1);
    i = 1;
  }

  if (digitalRead(buttonBpin) == LOW && p != 1) {
    sensorVal = (analogRead(analogSensor)/8);
    MIDI.sendNoteOn( 50, 127, 1);              // If "button2" is pressed, the note D is sended via channel 1.
    MIDI.sendControlChange(7, sensorVal, 1);   // Then, the volume (command 7) is adjusted to the "sensorVal" value.
    delay (2000);
    MIDI.sendNoteOff(50, 0, 1);
    p = 1;
  }
  if (digitalRead(buttonApin) == HIGH && i !=0) {
    i = 0;
  }
  if (digitalRead(buttonBpin) == HIGH && p !=0) {
    p = 0;
  }
}`

Hope you can help me.
Thanks in advanced.

franky47, 2020-04-19:
Does your synthesizer respond to ControlChange 7 sent from other sources ? If the notes are working, there's little chance for the CCs to not be working, so I'd look into CC assignment on the receiving side.

You can use MIDI OX on Windows or MIDI Monitor on macOS to see what the Arduino sends.

Imbr2, 2020-04-19:
I thought that the assignment was automatic but you were completely right, I configured the correct assignment in the synthesizer and now it works perfectly.
I really appreciate your help. Thank you.

closed 2020-04-19 by Imbr2