FortySevenEffects/arduino_midi_library #146: is it good for 3v3 input ?

fotosettore, 2020-04-20:
Hi !
i' trying to make a MIDI-IN interface. i'm using a 6N137 optoisolator
All works fine with MEGA2560 but ESP32 is deaf.
i tried a lot of hardware configurations and somewhere i read that library may be (i repeat : "may be") not ready for 3v3 signal.
waiting for delivery of new type of optoisolator (H11L1) i have a doubt.
is it right that library may be not ready for MIDI IN signal or i must to find solution in other way ?
i'm not skilled in this kind of installations and please excuse me about my post.
any help is appreciated

franky47, 2020-04-20:
The library works with hardware serial signals, if you manage to receive raw data, the library will plug on top of it and interpret the MIDI messages.

Now for your signal levels problem, if your microcontroller has a 3.3V-compatible UART, you should be able to power the optocoupler with 3.3V and have it convert the MIDI DIN current loop into 3.3V signals for the RX pin.

mink99, 2020-04-20:
See here: https://tttapa.github.io/Arduino/MIDI/Chap02-MIDI-Hardware.html

fotosettore, 2020-04-20:
many thanks for answer
after your post i think the problem is the type of signal given by optoisolator
i tried really a lot of configurations and i was wondered that there is not a tested schematic diagram in the web. many many solutions, like the one proposed by mink99 that at last id the original document released by midi.
Please note that ESP32 has only one (of three) free serial  so i used softwaresSerial to catch signals 
for MIDI OUT this kind of configuration works fine (in ESP32 too) but for MIDI IN it seems impossible.
i wait for H11L1 (new "name" of PC900) and i will update.

alf45tar, 2020-04-20:
May I suggest some tested solutions including schematic?
 
https://github.com/alf45tar/PedalinoMini#schematic

https://github.com/alf45tar/PedalinoMini/issues/12

https://github.com/FortySevenEffects/arduino_midi_library/pull/117

fotosettore, 2020-04-20:
> May I suggest some tested solutions including schematic? 

many thanks ... i study the diagram and the code and i will update

mink99, 2020-04-20:
The pedalino are 5 volt, we are looking for 3.3v

mink99, 2020-04-20:
And... dont use softwareSerial except for debugging...... the latency is far too high.....

fotosettore, 2020-04-21:
> May I suggest some tested solutions including schematic?
i tried the schematic and unfortunately it does not work. i want to study your code.

> And... dont use softwareSerial except for debugging...... the latency is far too high.....
So you think that the deaf situation of esp32 is latency of SoftwareSerial ?
However MIDI OUT works fine with SoftwareSerial. MIDI does not.
And SoftwareSerial seems the only way to have 2 serials on ESP32 (you know mega2560 has all HardwareSerial but for development needs i must use ESP32).
We all know that ESP32  actually has three hardware serials but at last only one is free to use.
I tried to use the hardwareserial for MIDI IN but but this too is deaf.
So : or there is a mistake in my breadboard using wrong optoisolator circuit or it is really a mystery how i can obtain a valid serial port to use :-(

fotosettore, 2020-04-21:
> Now for your signal levels problem, if your microcontroller has a 3.3V-compatible UART, you should be able to power the optocoupler with 3.3V and have it convert the MIDI DIN current loop into 3.3V signals for the RX pin.
My oscilloscope show 3.3 signal output ... 
This deafness if really very strange.

franky47, 2020-04-21:
Can I have a look at your code ?

fotosettore, 2020-04-21:
Here is my test.
First of all : i'm testing to put and get rig info to kemper profiler device.
(rig is the name of preset)

I'm using the following library :
[https://github.com/basarane/kemper-arduino-library](url)

but i modified the original kemper.cpp with a self customized, so it can works fine with esp32:
[http://www.homoandroidus.com/kemper/kemper.cpp](url)

and this is the sketch i'm using for test:
[http://www.homoandroidus.com/kemper/kemper_esp32.ino](url)

It sends every seconds a change of rig thorught MIDI OUT and display on screen the request of actual rig name arriving in MIDI IN.

This is the display shown if i use a mega2560 (tx1 - rx2) (also as only receiver for MIDI IN)
[http://www.homoandroidus.com/kemper/kemper_01.PNG](url)
This scenario works fine if i use HardwareSerial

And this is the display shown if i use a esp32 (tx26 - rx33)
[http://www.homoandroidus.com/kemper/kemper_02.PNG](url)
This scenario does not work, also if i use the HardwareSerial free as MIDI IN.

Please note that the optoisolator circuit is the same.
But overall note that if i send to MIDI OUT the request of change of kemper rig (using ESP32 with SoftwareSerial) and i connect the output of optoisolator to mega2560, all works fine !
request from esp32 and receive from mega -----> ok   (FIG. 1)
request from esp32 and receive esp32  ----> deaf     (FIG. 2)

franky47, 2020-04-21:
So the issue sounds more like a SoftwareSerial RX problem than a voltage issue, especially if you see 3.3V signals on the oscilloscope. Do you have interrupts enabled ? From what I remember of SoftwareSerial, they were needed to detect logic level changes and feed that to a shift register.

fotosettore, 2020-04-21:
Following your advice, I tried do disable the interrupts using command

`swSerIN.enableRx(false);`

and magically some data began to see themselfes on the arduino monitor !!!
But was a short victory because the data freezed after three or little more calls :-(
However, as you rightly said (many thanks for clue), the problem is in SoftwareSerial (EspSoftwareSerial).
I will ask for info to the author and i will post here the solution, if any (i really hope).
Last note : i tried 6N138 and TLP2361. No differences about data lost. Both has the same way to operate.
This is a comparation in my oscilloscope about the delay curves. 

http://www.homoandroidus.com/kemper/fotoaccoppiatori_grafici_midi.png

They seem not to affect operation.

franky47, 2020-04-22:
I don't think the problem is electrical, your signals have the right levels, and that slight delay will be ignored by discrete level logic, it sounds indeed like a software issue. Let us know your findings !

closed 2020-04-22 by franky47