FortySevenEffects/arduino_midi_library #147: Version 5.0.0 "reflects" MIDI messages over USB

SteveCooling, 2020-04-22:
Just tried to move a synth project over to versjon 5.0.0. 

I have Arduino Leonardo (clone.. a Pro micro) connected via USB. It is supposed to take MIDI messages and convert them to CV/Gate ++ for an analog synth.

Compiling using Arduino 1.8.12 on macOS 10.15.4

Two failures occur on version 5.0.0 (confirmed not a problem on version 4.3.1)

1. NoteOn and NoteOff messages are "reflected" back via USB, but with incorrect note and velocity.
2. The NoteOn/NoteOff handler callbacks in the firmware are being fred the same bad data.

See attached .ino example and screenshot of MIDI messages passed.

```c
#include <USB-MIDI.h>

USBMIDI_CREATE_DEFAULT_INSTANCE();

unsigned long t0 = millis();

#define PIN_CV 3

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);

  // Listen for MIDI messages on channel 1
  MIDI.begin(1);

  MIDI.setHandleNoteOn(OnMidiNoteOn);
  MIDI.setHandleNoteOff(OnMidiNoteOff);

  pinMode(PIN_CV, OUTPUT);
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
void loop()
{
  // Listen to incoming notes
  MIDI.read();

}

static int noteToPWM(int note) {
  return note * 2;
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
static void OnMidiNoteOn(byte channel, byte note, byte velocity) {
  analogWrite(PIN_CV, noteToPWM(note));
}

// -----------------------------------------------------------------------------
//
// -----------------------------------------------------------------------------
static void OnMidiNoteOff(byte channel, byte note, byte velocity) {
  //analogWrite(PIN_CV, 0);
}
```

[image]

franky47, 2020-04-22:
Thanks for the report, @lathoub can you have a look at the incorrect data being passed to the callbacks ?

As for the reflection, it might be due to SoftThru being enabled by default, while this makes sense for hardware MIDI, it does not for USB and probably other transports.

lathoub, 2020-04-22:
Hi, I tried to reproduce the issue, but can't.. I have used the updated `NoteOnOffEverySec.ino` demo.

Here is my setup: Leonardo with Sparkfun MIDI shield, the shield is connected to an M-Audio MIDI USB device.

- Using 'Pocket MIDI' I send MIDI notes and they are received correctly by the Leonardo (notes, velocity and channel match)
- The Leonardo sends notes over the M-Audio and they are received correctly in 'Pocket MIDI' (in the MIDI In Monitor)

Not sure what is happening

lathoub, 2020-04-22:
@SteveCooling have you cloned the repo or used the IDE to download the USBMIDI library?

lathoub, 2020-04-22:
I have issued a new release of the USB-MIDI library (v1.1.0)

lathoub, 2020-04-23:
The Arduino has picked up on the new release (v1.1.0) and should be fine now (I forgot to make a Release version of USB-MIDI when 5.0.0 of the MIDI Lib came out 🙈).
@SteveCooling please test again

SteveCooling, 2020-04-23:
Tested and works fine.
Version 1.1.0 of USB-MIDI got rid of the corrupted data.
`MIDI.turnThruOff();` removed the unexpected "reflection".
👍

closed 2020-04-23 by SteveCooling

franky47, 2020-04-23:
Thanks, we might need to disable Thru for non-serial transports then.

SteveCooling, 2020-04-23:
Yes it was very unexpected behaviour, and created a loop when I routed MIDI traffic from Ableton to it 😄

lathoub, 2020-04-23:
How about adding a method in the Transport layer:

```
...
bool thruActivated()
{
  return true;
}
...
```

in MIDI.hpp

```
template<class Transport, class Settings, class Platform>
void MidiInterface<Transport, Settings, Platform>::begin(Channel inChannel)
{
...
    mMessage.data2   = 0;
    mMessage.length  = 0;

    mThruFilterMode = Thru::Full;
    mThruActivated  = mTransport.thruActivated();
}

```

franky47, 2020-04-23:
Good idea, it could be a static bool member rather than a function though.

lathoub, 2020-04-23:
Ah, good!

`static const bool thruActivated = false;
`
```
...
    mThruFilterMode = Thru::Full;
    mThruActivated  = mTransport.thruActivated;
}
```

lathoub, 2020-04-23:
pr #148